cra-ros-pkg/robot_localization #352: Integration RSSI Localization

reinzor, 2017-03-06:
Hi,

I was thinking of integrating RSSI measurements (indoor instead of GPS) with this package; does anybody have experience with this? I've found some papers online and these approaches seem to be accurate up to ~1 meter. 

I see now some options for the integration

- Fusing RSSI measurements into a sensor_msgs/NavSatFix and convert it using the navsat_transform_node to odom
- Fusing RSSI measurement to nav_msgs/Odometry directly
- Integrating RSSI measurements into the filter..

Let me know if you have any experience or thoughts.

AlexisTM, 2017-03-08:
@reinzor I am going to use a UWB localization and I will be in the same position as you soon. I think you send your data as an Odometry message and sending a covariance of 99999 for elements not computed.

AlexisTM, 2017-03-08:
@reinzor Check that video! Awesome information :D 

http://www.ros.org/news/2016/06/tom-moore-working-with-the-robot-localization-package.html

reinzor, 2017-03-08:
Hi @AlexisTM , thanks I know that video :). But looking back at this, I am wondering how to calibrate the RSSI position odometry with the base odometry. In the navsat_transform node this is explicitly done so that the gps odom and wheel odom have the same starting point.

eroniki, 2018-01-25:
@reinzor I have implemented a similar localization scheme. My RSSI localizer provides a global location, not the pose, by using trilateration method. In my case, the resulting estimations are converted [PoseWithCovarianceStamped](http://docs.ros.org/api/geometry_msgs/html/msg/PoseWithCovarianceStamped.html) messages. Therefore, the robot_localization package was able to use the estimations as sensor measurements. It worked well for my case and I would recommed using in the same scheme.

reinzor, 2018-01-26:
Hi Murat, thanks for the information! We are planning the same approach.

-Rein