cra-ros-pkg/robot_localization #354: Precision loss reading datum from paramater in navsat_transform.cpp

Kaiser1401, 2017-03-20:
The code used to read the manual datum from the config as set via the yaml-file causes a precision loss.
E.g. a lattitude of  57.8306145 becomes 57.8306.
Combining this nodes with other nodes using GPS data converted to UTM I found an offset of about 2m and could track it down to navsat_transform.cpp lines 127ff:
```c++
std::ostringstream ostr;
ostr << datum_config[0] << " " << datum_config[1] << " " << datum_config[2];
std::istringstream istr(ostr.str());
istr >> datum_lat >> datum_lon >> datum_yaw;
```

we could use the following instead:
```c++
       if (datum_config[0].getType()==XmlRpc::XmlRpcValue::TypeDouble)
        {
          datum_lat = (datum_config[0]);
        }
        if (datum_config[0].getType()==XmlRpc::XmlRpcValue::TypeInt)
        {
          datum_lat = (int)(datum_config[0]);
        }

        if (datum_config[1].getType()==XmlRpc::XmlRpcValue::TypeDouble)
        {
          datum_lon = (datum_config[1]);
        }
        if (datum_config[1].getType()==XmlRpc::XmlRpcValue::TypeInt)
        {
          datum_lon = (int)(datum_config[1]);
        }

        if (datum_config[2].getType()==XmlRpc::XmlRpcValue::TypeDouble)
        {
          datum_yaw = (datum_config[2]);
        }
        if (datum_config[2].getType()==XmlRpc::XmlRpcValue::TypeInt)
        {
          datum_yaw = (int)(datum_config[2]);
        }
```

(not so experienced with git regarding creating pull requests etc... please excuse)

ayrton04, 2017-03-20:
Please see https://github.com/cra-ros-pkg/robot_localization/issues/338.

ayrton04, 2017-03-20:
If you are feeling adventurous, you could try adding a std::setprecision(10) before the stream operators, e.g.,

```
std::ostringstream ostr;
ostr << std::setprecision(10) << datum_config[0] << " " << datum_config[1] << " " << datum_config[2];
std::istringstream istr(ostr.str());
istr >> datum_lat >> datum_lon >> datum_yaw;
```

closed 2017-03-20 by ayrton04

Kaiser1401, 2017-03-21:
`std::setprecision(10)` works fine!
Thats an easier solution than mine.
Sorry for the duplicate, must have overlooked it.

ayrton04, 2017-03-21:
Any chance you could submit a PR?