cra-ros-pkg/robot_localization #357: Set initial state via parameters

ayrton04, 2017-03-28:
We should enable users to set the filter's initial state via a parameter.

ayrton04, 2017-08-09:
Fixed in https://github.com/cra-ros-pkg/robot_localization/pull/358.

closed 2017-08-09 by ayrton04

siddharthbhurat4, 2023-05-23:
I am trying to set this ros parameter from my python file where I record my initial offsets.
When I check that parameter using rosparam get command from command line, it is giving the correct values, but the filter is not starting at that position. I am using ekf.
This is how I am setting it:

`            rospy.set_param('/initial_state', [x_offset, y_offset, 0.0, 
                                               0.0, 0.0, th_offset,
                                               0.0, 0.0, 0.0,
                                               0.0, 0.0, 0.0,
                                               0.0, 0.0, 0.0])`

ayrton04, 2023-05-25:
@siddharthbhurat4 please ask questions on answers.ros.org.

siddharthbhurat4, 2023-05-25:
Hey Tom, I have asked a question on ros answers. Please can you guide me there. 
The name of post is - robot_localization package ros

siddharthbhurat4, 2023-05-26:
I am having a confusion about the transformations which I should have when using the given config of sensor. Please can you guide me there.

ayrton04, 2023-06-05:
Can you please just provide a link to the question?