cra-ros-pkg/robot_localization #362: Feature Request : Support for 9 dimensional Navigation state

kshitijgoel007, 2017-04-06:
Hello everyone,

Thanks for maintaining this package. I wish to know if this only works for a 15-dimensional state only? If yes,  can we modify the YAML in order to obtain a 9-dimensional setup?

I am trying to use this package for use in an underwater vehicle. My navigation state vector is  : [x y z theta psi phi u v w]

Awaiting your response!

Best,
Kshitij

ayrton04, 2017-04-06:
Is there a [ROS answers question](http://answers.ros.org/questions/) for this? If so, can you please point me to it?

Just so we're clear, what does the vector `[u v w]` represent in this case?

As it is, the filter tracks a 15-D state of the vehicle, but that doesn't mean you have to use all of the information. Fuse what data you have and read out what you need. The output is a nav_msgs/Odometry message, which contains pose and velocity.

ayrton04, 2017-05-03:
Can this be closed, or are you still requiring functionality that the package lacks?

ayrton04, 2017-09-24:
Closing. Feel free to discuss further and I will re-open if needed.

closed 2017-09-24 by ayrton04