cra-ros-pkg/robot_localization #363: ImportError: No module named defusedxml.xmlrpc

DaDaDoDoLee, 2017-04-26:
# When I executed the command: 'roslaunch ekf_template.launch' in a terminal,I got several warning as follows:
Traceback (most recent call last):
  File "/home/pantiq/robot/devel/bin/roslaunch", line 6, in <module>
    exec(fh.read())
  File "<string>", line 34, in <module>
  File "/home/pantiq/robot/devel/lib/python2.7/dist-packages/roslaunch/__init__.py", line 35, in <module>
    exec(__fh.read())
  File "<string>", line 56, in <module>
  File "/home/pantiq/robot/src/ros_comm/tools/roslaunch/src/roslaunch/launch.py", line 55, in <module>
    from roslaunch.nodeprocess import create_master_process, create_node_process
  File "/home/pantiq/robot/src/ros_comm/tools/roslaunch/src/roslaunch/nodeprocess.py", line 52, in <module>
    from rosmaster.master_api import NUM_WORKERS
  File "/home/pantiq/robot/src/ros_comm/tools/rosmaster/src/rosmaster/__init__.py", line 35, in <module>
    from .main import rosmaster_main
  File "/home/pantiq/robot/src/ros_comm/tools/rosmaster/src/rosmaster/main.py", line 43, in <module>
    import rosmaster.master
  File "/home/pantiq/robot/src/ros_comm/tools/rosmaster/src/rosmaster/master.py", line 47, in <module>
    import rosmaster.master_api
  File "/home/pantiq/robot/src/ros_comm/tools/rosmaster/src/rosmaster/master_api.py", line 72, in <module>
    from rosmaster.util import xmlrpcapi
  File "/home/pantiq/robot/src/ros_comm/tools/rosmaster/src/rosmaster/util.py", line 48, in <module>
    from defusedxml.xmlrpc import monkey_patch
ImportError: No module named defusedxml.xmlrpc
# Then I downloaded a folder named defusedxml from a website,and put it in the robot_localization/src path,still it doesn't work.
What should I do with it?
I do appreciate it if anyone can help me out.

DaDaDoDoLee, 2017-04-27:
Thank you for Ahendrix's answer at ROS FAQ.He told me to run sudo apt-get install python-defusedxml in a terminal and it really works.
But here pops up another problem: After running the command: roslaunch ekf_template.launch , I run rostopic echo /odometry/filtered in another terminal.Then I got nothing.Does it mean that the node isn't operating at all or something else goes wrong?

ayrton04, 2017-04-27:
Please see the other issue you posted.

closed 2017-04-27 by ayrton04