cra-ros-pkg/robot_localization #364: pub no data after running roslaunch

DaDaDoDoLee, 2017-04-27:
With using robot_localization offered by ROS downloaded from github,I came up with a problem: I run roslaunch ekf_template.launch in a terminal and it turns out to be good.Then I run rostopic echo /odometry/filtered in another terminal while rosbag play a bagfile in another terminal.But I got nothing.Does it mean that the node isn't operating at all or something else goes wrong?

ayrton04, 2017-04-27:
The template file is just there to show you how to configure the node. It's not intended to actually produce any data. You have to have sensor data going into the node in order for it to work (so, e.g., the `odom0` topic has to have data being published on it).

I would recommend a thorough read-through of the packages's [documentation](http://docs.ros.org/kinetic/api/robot_localization/html/index.html).

closed 2017-04-27 by ayrton04

DaDaDoDoLee, 2017-04-27:
Thank you so much Tom,I'll do a thorough research on the documentation you recommended.
But there's pulisher publishing /odom/filtered topic right in the function run().
it goes like this:positionPub.publish(filteredPosition) to fire off the position and the transform.Actually I got confused by it.

ayrton04, 2017-04-27:
That publication happens only if the filter has had any sensor data input. I'm not sure how new you are to ROS, but each node can subscribe to topics with data that it needs, and publish to topics itself. In this case, the filter will not publish anything if there is no sensor data. This is intended, as it is an indication to the user that the data it needs are not present.

Four useful tools: `rosnode list` and `rosnode info` for seeing what nodes are present and what each node is subscribing and publishing to, and `rostopic list` and `rostopic info` to see what topics are present and what is publishing and subscribing to each topic.

DaDaDoDoLee, 2017-04-27:
I do play a bagfile named test2.bag in another terminal.It seems that the bagfile performs well so can I assume that there's no problem with sensor data input?
I run rosnode list,it turns out to be like this:
/ekf_se
/play_1493281205256917011
/rosout
/rostopic_8444_1493281180577
/rqt_gui_py_node_7325
As presented above,/rostopic_8444_1493281180577 is the subscriber of topic /odometry/filtered.
Sorry about bothering you again Tom,I have no idea where the crux of the problem lies in.

ayrton04, 2017-04-27:
The EKF [template parameter file](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L158) is subscribing to the following topics:

- [example/odom](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L73)
- [example/another_odom](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L123)
- [example/pose](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L136)
- [example/twist](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L148)
- [example/imu](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml#L158)

*None of those topics exist in the test2.bag file.* Here is the `rosbag info` output for that bag:

```
$ rosbag info test2.bag 
path:         test2.bag
version:      2.0
duration:     1:01s (61s)
start:        Jan 01 1970 01:00:47.58 (47.58)
end:          Jan 01 1970 01:01:48.78 (108.78)
size:         521.2 KB
messages:     6121
compression:  bz2 [5/5 chunks; 13.01%]
uncompressed:   3.3 MB @ 55.1 KB/s
compressed:   438.6 KB @  7.2 KB/s (13.01%)
types:        nav_msgs/Odometry [cd5e73d190d741a2f92e81eda573aca7]
              sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:       /imu/data                          3061 msgs    : sensor_msgs/Imu  
              /jackal_velocity_controller/odom   3060 msgs    : nav_msgs/Odometry
```

Therefore, the node will not run, because it is not getting an sensor data. The EKF doesn't automatically "find" the sensor data; you need to tell it _what_ sensor data (topics) to listen for.

Again, the template parameter file was _not_ meant to be used. It is an example that is meant to show you how to configure the node. You need to tell the node to subscribe to the topics that are in the bag. I would perhaps work through the ROS tutorials again, just to get a firm understanding of how topics and publishing/subscribing work.

DaDaDoDoLee, 2017-04-27:
Sweet!I renamed example/odom to /jackal_velocity_controller/odom and example/imu to /imu/data as suggested in the bagfile,then run rostopic echo /odometry/filtered in a terminal,the desired data of  /odometry/filtered pop out.
Am I doing it in the right way?Does it mean I can use robot_localization directly ,all I need is to adjust the name of sensor data?
So nice of you !

ayrton04, 2017-04-27:
Not directly, no. You need to modify many of the other parameters, especially the config parameters (e.g., `odom0_config`). The documentation goes over all of that in explicit detail. Good luck!

DaDaDoDoLee, 2017-04-27:
Thank you so so so much!!!