cra-ros-pkg/robot_localization #366: What do three frames(baselink\odom\map) mean?

DaDaDoDoLee, 2017-05-03:
I've read over robot_localization documents , REP105 and the ROSCon 2015 presenation but still got confused by these frames. Here's my question:

1.About base_link frame: It's said to be attached to the robot.Does it mean it always keeps x axis pointing forward like ENU coordinates and move with the robot?

2.About odom frame: Why is it continuous ? The input data of imu and odometry are offered in certain frequency,isn't it discrete data? And why does it drift over time?Is it because the position is calculated by twiste data rather than given by GPS?

3.About map frame: As can be seen from the rviz, the map frame seems to go with odom frame.What's the diffrence between them?

Last but not least,my GPS,IMU,Odometry all pub discrete data but vary in frequency. What I learn from the robot_localization package is that IMU and Odometry are continuous.Which is right? I do appreciate it if someone can enlighten it to me.

ayrton04, 2017-05-03:
Hi,

This board is for reporting bugs and feature requests for the software. Please post questions on answers.ros.org, and I will answer when I get some time.

closed 2017-05-03 by ayrton04

DaDaDoDoLee, 2017-05-03:
Ok,I already posted questions on ROS answers, looking forward to your answer.
thank you so much

DaDaDoDoLee, 2017-05-04:
Hey Tom. Sorry for bothering you again. I have several question on customizing the robot_localization. Here's my question:

1. All I need to modify are ekf_template.yaml and transfer function matrix in ekf.cpp, is that so?

2. What should I do with those covariances(both P and Q) of unfused variables,setting it to a small value(eg.1e-12) or inflate it to large value (eg.1e3)?

3. Is transfer function matrix in ekf.cpp using right-handed coordinate system affixed to the robot with X axis pointing forward?

I do appreciate it if you can give me some tips. Thank you so much!

ayrton04, 2017-05-04:
Again, please use ROS Answers. There are [a lot of questions](http://answers.ros.org/questions/scope:all/sort:activity-desc/tags:robot_localization/page:1/) for me to get through, so it may take me some time. I would not recommend modifying ekf.cpp. If you find the motion model doesn't work for you, then you should create a new class that has your motion model.

DaDaDoDoLee, 2017-05-04:
Ok...Thanks a lot