cra-ros-pkg/robot_localization #367: How do the robot_localization solve the multi odometry?

echotor, 2017-06-13:
If I have multiply odometry source, and I use the robot_localization, how does it solve these data?

If I config both true in x,y,z in all source, which one will can affect? (If the value of x,y,z in all source is a little different.)

AlexisTM, 2017-06-18:
The most important thing is to have coherent data. Ensure that all the TF transformations are valid.

For multiple odometry, you can specify odom0, odom1, .. (See the examples) and make sure that you have only one absolute measurement. If you have multiple position measurements, use the "odomx_differential" to true to ensure your measures do not jump. It will transform the position to velocities. 

_Sent from my UMI TOUCH using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

ayrton04, 2017-06-21:
I'm assuming there is a ROS Answers question somewhere for this, correct? Can you point me to it? I have been neglecting r_l lately due to work, but I will do my best to get back to answering questions and actually supporting the package soon. :)

echotor, 2017-06-22:
The odometry data from multi sources are maybe different. So I can not find a solve for handling them. I think the point is that which one will do work while we get multi data. So maybe it needs other sonar to support. Do you think so?

ayrton04, 2017-06-22:
If the data sources differ, then:

(a) Fuse your best source as normal,
(b) Turn on `differential` mode for the others

AlexisTM, 2017-06-22:
I think you will need to give more informations about your setup.

Those are 3 good material to start with: 

* General explaination of the package (from @ayrton04) : https://vimeo.com/142624091
* The API documentation: http://docs.ros.org/lunar/api/robot_localization/html/configuring_robot_localization.html
* The configuration template file: https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/params/ekf_template.yaml

ayrton04, 2017-06-28:
If you have asked this on ROS Answers (answers.ros.org), can you please point me to it? This board is meant for bug reports, so I will close this ticket.

closed 2017-06-28 by ayrton04