cra-ros-pkg/robot_localization #368: release for lunar

anthony-jclark, 2017-06-14:
Currently I have robot_localization installed form source for my project, which necessarily uses lunar. Before I give my students some instructions on how to install robot_localization from source I just wanted to ask if you have an estimate on when you'll prepare a release for lunar that will be indexed and available to be installed via a package manager.

Thanks, and sorry for the question, but I'm not entirely familiar with the process of releasing a catkin package.

ayrton04, 2017-06-14:
Hi! The delay is entirely down to me just not getting my act together. I will try to get it released by tomorrow, though the package servers won't pick up on it until they do another lunar sync. Apologies for the lag!

anthony-jclark, 2017-06-14:
Thank you for the quick reply. No need for apologies.

anthony-jclark, 2017-06-16:
The package has already been synced into the main public ros repository.

closed 2017-06-16 by anthony-jclark

ayrton04, 2017-06-16:
Yeah, managed to sneak that one in just in time!