cra-ros-pkg/robot_localization #369: BUG - catkin_make - Ubuntu mate - RapsberryPi 3

betterRobotics, 2017-06-19:
NEVER SUBMITTED A BUG BEFORE!! 

I installed this package on my RPI3 with ubuntu mate, it seems to be hanging when building the ros_fitler.cpp. when pressing crtl+c it responds like it's stuck in a while loop which blocking the interrupt key command.

ayrton04, 2017-06-21:
Er, that sounds a bit like a compiler issue. I don't have a Raspberry Pi, so it's hard for me to replicate, but I have built the package successfully on ARM boards before. Can you use catkin_tools instead and run `catkin build`?

tonybaltovski, 2017-06-21:
Are you running out of RAM and swap?

betterRobotics, 2017-06-22:
I am not sure how to check that, but the PI3 has 1G of RAM so I wouldn't think so. I was like the the makefile was waiting for something to build it would move on, each time I killed it I saw something like pr.wait_for() in some of the error code. I'll try catkin build when I have another chance to play with it got heaps of uni work to finish and I've implemented robot_pose_ekf for now, this package looks a lot more flexible and documentation is really nice so I will try to get it running on the PI.

Cheers Ben.

tonybaltovski, 2017-06-22:
Run `top` while compiling and watch the amount of memory and swap that is free.

caseypen, 2017-09-20:
Hi All,
Are there any updates about this issue? I met the same problem with Cheers and it stopped at the compiling process "Building CXX object robot_Localization/CMakeFiles/ros_robot_localization_listener.dir/src/ros_robot_localization_listener.cpp.o". I checked top and it runs out of memory while doing compiling. 
Chen Peng

ayrton04, 2017-09-24:
I don't have any of the boards in question in my possession, I'm afraid, so I can't replicate the issue. Can you try cloning and old version of the software from one of the release tags and building that? It would at least help to isolate when this became an issue. It's interesting that these build issues appear to be isolated to certain compilers.

tonybaltovski, 2017-09-24:
@caseypen do you have swap on your Raspberry Pi 3? And if so, how much do you have?

caseypen, 2017-09-24:
Hi All,
Thanks for replying. @tonybaltovski the swap size is set as default, it should be 100M. 
The compiling of latest version is finished by several times of breaking and restarting. It can work now.
Thanks!
Chen

ayrton04, 2017-12-07:
Can this be closed?

BetterRobotics, 2017-12-07:
Hi Tom,

I still feel the problem could be rectified for ARM based projects but, I was unable to determine if it was indeed the swap running out due to time constraints. It would be nice if it was able to build on the ARM processor! But unfortunately I am unable to assist with correcting the issue however I can help with testing.

caseypen, 2017-12-08:
Hi Tom,
Thanks for your help and replying. Actually I finishing compiling on
Raspberry Pi though with several times of out of memory error. But I think
it should work fine on other ARM based board with better configuration.
Best

Chen

ayrton04, 2018-06-19:
Have you tried compiling with `catkin build -j1`? I am going to close, but if anyone has time/interest, feel free to reopen.

BetterRobotics, 2018-06-19:
Hi Tom,

As Chen mention it was in fact a memory error the pi is running out of memory before it can complete the entire build! Chen’s method of restart/build didn’t work for me. Unsure to why!

I have tried -j1 command but this didn’t not work as it’s not an over processing issue! It’s memory! From memory (going back a bit) I always failed at about 70% from a “fresh build” but never had success to actually build a working package! I was using Ubuntu mate 16.04 on pi3.

Don’t think this should be closed as you need more people to provide testing! I might be able to help out in the next few weeks but need to know exactly what you need for debug to get rid of the ninja bugs! ☺️

ayrton04, 2018-06-19:
Yes, understood that it's a memory issue, but `-j1` forces a single thread, and so if we were compiling two or more source files at the same time before, compiling one should use less. Regardless, I can keep this open.

BetterRobotics, 2018-06-19:
Agreed, I feel the the -j1 option might have lead to Chan’s success but cannot stipulate any further, being a single thread means that when it runs out of SWAP memory the compiler and locks up the logging feature has a greater chance of actually resuming the build process in at the correct location. I will have a bit of time over the next few weeks to test it out. I’m not super linux orientated but can follow instruction to nail the issue if your interest to resolve this.

Cheers Ben

subodh-malgonde, 2018-09-04:
`catkin_make -j1` worked for me. It took a long time for the code to build though.

tomsepe, 2019-03-28:
I also ran into this issue today. Where catkin_make just hung.

Trying the catkin_make -j1 option and so far it seems to be working.... I got based the hang point (see below) and is now building fine, although slowly :)

[image]

SteveMacenski, 2019-12-04:
I built this on my jetson nano over the weekend without a problem (A57 cores) with my catkin config. I'm inclined to say this ticket should be closed since its just requiring swap.

BetterRobotics, 2019-12-05:
Yes close ticket, just swap issue no issues with actual build 👌🏼 thanks all
👌🏼

closed 2019-12-05 by SteveMacenski