cra-ros-pkg/robot_localization #370: IMU differentiation incorrect

jpapon, 2017-06-23:
Hello,

So I've been struggling to get my configuration working properly. In my current setup, I have the following:
imu0 - Xsens IMU reporting at 400hz
odom0 - Wheel odometry reporting twists
pose0 - LIDAR localization, reporting absolute poses

I've found that the Xsens IMU absolute orientation is far more accurate (due to internal filtering) than using the IMU angular velocities directly. So I'd like to use it.

I've set up my configuration using two EKFs, as below. Unfortunately, this does not work, because the absolute orientation reported by the IMU is not aligned to the map - the IMU resets its heading whenever it restarts, so it is not aligned to the map that pose0 is using.

To get around this, I tried using the imu0 in differential mode, but it does not work - the orientations I get from doing this are completely incorrect - it seems to rotate far too much (ie 180+ degrees for a 90 degree turn).

I can use the output of the first EKF as the input to the second one instead of imu0 and odom0, and differentiate, and this works. It's not ideal though I think. 
I've attached a bag and urdf file as well. The URDF just provides imu -> base_link.
In the attached bag file, the IMU orientation happens to be 90 degrees off the pose0 orientation, but that's just coincidence. The orientation of the IMU zeroes to whatever heading it starts on.
[config.zip](https://github.com/cra-ros-pkg/robot_localization/files/1098806/config.zip)

To test things, I've been setting the pose0 input to "relayed_lidar_pose" and using a relay to turn it on and off:
rosrun topic_tools relay /estimated_lidar_pose /relayed_lidar_pose

Any help would be appreciated! Is imu differentiation just broken? Or is it because I'm running the imu at 400hz, and differentiating it at 40hz just accumulates too much error?

```
ekf_se_imu_odom:
  frequency: 40
  sensor_timeout: 0.025
  two_d_mode: true
  transform_time_offset: 0.0
  print_diagnostics: true
  debug: false
  debug_out_file: /path/to/debug/file.txt
  publish_tf: true
  publish_acceleration: false

  map_frame: kyle_map              # Defaults to "map" if unspecified
  odom_frame: odom            # Defaults to "odom" if unspecified
  base_link_frame: base_link  # Defaults to "base_link" if unspecified
  world_frame: odom           # Defaults to the value of odom_frame if unspecified

  odom0: estimated_encoder_pose
  odom0_config: [false,  false,  false,
                 false, false, false,
                 true,  true,  false,
                 false, false, true,
                 false, false, false]
  odom0_differential: false
  odom0_queue_size: 10
  odom0_nodelay: false
  odom0_relative: false

  imu0: imu/data
  imu0_config: [false, false, false,
                false,  false,  true,
                false, false, false,
                false, false, false,
                true,  true,  false]
  imu0_nodelay: false
  imu0_differential: false
  imu0_relative: false
  imu0_queue_size: 10
  imu0_remove_gravitational_acceleration: true

ekf_se_map:
  frequency: 40
  sensor_timeout: 0.025
  two_d_mode: true
  transform_time_offset: 0.0
  print_diagnostics: true
  debug: false
  publish_tf: true
  publish_acceleration: true

  map_frame: kyle_map
  odom_frame: odom
  base_link_frame: base_link
  world_frame: kyle_map

  odom0: estimated_encoder_pose
  odom0_config: [false,  false,  false,
                 false, false, false,
                 true,  true,  false,
                 false, false, true,
                 false, false, false]
  odom0_differential: false
  odom0_queue_size: 10
  odom0_nodelay: false
  odom0_relative: false

  imu0: imu/data
  imu0_config: [false, false, false,
                false,  false,  true,
                false, false, false,
                false, false, false,
                true,  false,  false]
  imu0_nodelay: false
  imu0_differential: false #Setting this to true breaks everything.
  imu0_relative: false
  imu0_queue_size: 10
  imu0_remove_gravitational_acceleration: true

  # This works, but probably isn't ideal.
  # odom0: odometry_IMU_encoders
  # odom0_config: [true,  true,  false,
  #                false, false, true,
  #                false,  false,  false,
  #                false, false, false,
  #                false, false, false]
  # odom0_differential: true
  # odom0_queue_size: 10
  # odom0_nodelay: false
  # odom0_relative: false

  pose0: estimated_lidar_pose
  pose0_config: [true, true,  false,
                 false, false, true,
                 false, false, false,
                 false, false, false,
                 false, false, false]
  pose0_queue_size: 10
  pose0_nodelay: false
  pose0_differential: false
  pose0_relative: false
```

Launch file:
```
<node pkg="robot_localization" type="ekf_localization_node" name="ekf_se_imu_odom" clear_params="true" output="screen">
	  <remap from="odometry/filtered" to="odometry_IMU_encoders"/>  
    <remap from="set_pose" to="/ekf_se_imu_odom/set_pose"/>  
  </node>

  <node pkg="robot_localization" type="ekf_localization_node" name="ekf_se_map" clear_params="true" output="screen">
      <remap from="odometry/filtered" to="filtered_odometry"/>
      <remap from="set_pose" to="/ekf_se_map/set_pose"/>  
  </node>

  <param name="jake_description" command="$(find xacro)/xacro '$(find jake_launch)/config/jake.urdf.xacro' --inorder
    robot_name:=jake" />

  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher">
        <remap from="robot_description" to="jake_description" />
  </node>
```

ayrton04, 2017-06-26:
Is there a ROS Answers question for this? If so, can you point me to it?

jpapon, 2017-06-26:
No, but I can post it there if you'd like. 

Mainly I'd like to know if it's possible to get good results running an IMU in differential mode?

ayrton04, 2017-06-26:
It should be, yes. I've done it before. This kind of thing typically comes down to a frame issue, but it's hard to say without sample messages from each input source and bags. If you can please post it on ROS Answers, that would be great. If we find it is a bug, I can address it here.

jpapon, 2017-06-26:
There's a bag attached to my original post.

The IMU orientations are definitely in a different frame than the laser scanner orientations - the laser scanner is relative to map, and the IMU is relative to heading at start-up (so, odom, I suppose).  As I said, it works okay if I use the velocities directly, or if I use the velocities output of my odom+imu filter as the input to my second EKF. It just does not work at all if I try to differentiate my IMU orientations - the angular velocities output by differentiation are far too high, and have the robot spinning all over the place.

ayrton04, 2017-07-31:
OK, so I had a chance to look into this. The issue appears to be your IMU's time stamps. To test, I did the following:

1. I echoed the IMU data to a plotting-friendly file: `rostopic echo /imu/data -p > imu0.txt`
2. I manually exported the computed differential yaw velocity and its corresponding timestamp to a file:

```
if (topicName.find("imu") != std::string::npos)
{
  std::cerr << std::setprecision(20) << ros::Time::now().toSec() << ", " << yawVel << "\n";
}
```

(I just redirected the error output to a file).

I then did the following:

1. Converted the IMU's orientation quaternion to Euler angles
2. Computed the velocity manually via Octave/MATLAB: `yaw_vel_diff_imu = diff(imu_yaw) ./ diff(imu_time)`
3. Plotted the EKF's computed differential velocity:

[image]

4. Overlaid the manual values from (2):

[image]

Note that the manually-computed differential velocity overlays the EKF's computed value. The EKF has lower frequency, but the same magnitude:

[image]

5. I then overlaid the yaw velocity as determined by your IMU's gyros:

[image]

  Note the differentiated values have a strong biases, thus over-estimating your true velocity during the turns.

6. Just for completeness, I did a simple plot of the differentiated time stamps *only* from your IMU data. Here's what it looks like:

[image]

So, it appears that something is amiss with your IMU time stamps. The fact that manual differentiation produces the same output as the EKF is telling. Also, you can clearly see that your IMU time stamps are not regular: the deltas between them oscillate between ~0.001 (1000 Hz) and ~0.004 (250 Hz), but seem to slowly converge over time towards ~0.0025 (400 Hz).

Each of those measurements will have the same covariance associated with it, so the filter is going to have a yaw velocity estimate that is biased and over-estimated in the same manner as the differentiated values.

ayrton04, 2017-07-31:
Oh, I should also note that once the time stamp deltas start to converge, they suddenly get worse again, and then repeat their convergence:

[image]

jpapon, 2017-07-31:
Wow, thank you for doing this analysis, this was very helpful!!
It appears the issue is coming from ethzasl_xsens_driver, which is publishing the imu data messages stamped with rospy.Time.now(). This makes the timestamp delta dependent on CPU load from other nodes.
That this is happening is a bit puzzling to me; this is a pretty widely used package, so this seems like a pretty big oversight.

jpapon, 2017-08-01:
So there's a time reference contained in the raw message coming from the Xsens IMU. I used it to calculate an initial offset, and then I used that as the time instead of rospy.Time.now(). 

The EKF now works much much much better - both with and without differential mode. I've pushed my changes to my fork: https://github.com/jpapon/ethzasl_xsens_driver in case anyone else needs them.

Thanks so much for your help @ayrton04 , you showed me the light! Closing this issue now!

closed 2017-08-01 by jpapon

ayrton04, 2017-08-01:
You're welcome. I wonder if it's worth PR-ing that change to the main repo, and maybe making it a parameter. Good luck!