cra-ros-pkg/robot_localization #371: Use of pose from a unique odometry message in both ekf_se_odom and ekd_se_map

Yvonnn, 2017-06-27:
Hello,

I am fusing odometry from a lidar (via undocumented pub_odometry publish parameter of hector_slam), IMU and GPS data. I am using the dual_ekf_navsat_example.
My lidar odometry has no twist, only pose. So that it is taken into account by ekf_se_odom, I need to publish it with frame_id = odom, because world_map of ekf_se_odom is odom.
Then, if I want my lidar odometry to be fused again in ekf_se_map, I need to publish it with frame_id = map, because world_map of ekf_se_map is map.
So... I guess I have to duplicate this odometry... am I right or am I missing something ?

ayrton04, 2017-06-28:
Did you ask this on answers.ros.org? If so, can you please point me to it?

Yvonnn, 2017-06-29:
Sorry about that. It is done here now : http://answers.ros.org/question/265007/what-is-good-practice-to-fuse-pose-from-a-unique-odom-message-in-robot_localization-in-both-ekf_se_odom-and-ekf_se_map/?comment=265077#post-id-265077

ayrton04, 2017-09-24:
Closing (question answered).

closed 2017-09-24 by ayrton04