cra-ros-pkg/robot_localization #374: navsat_transform_node: Unclear direction of parameter magnetic_declination_radians

nxdefiant, 2017-07-09:
I check the the linked source for magnetic orientation with two cities:

1. Germany/Hamburg: Declination is 2° 44' E
2. U.S.A./New York: Declination is 12° 58' W

the documentation did not specify the direction here, so after checking the source code I assume to enter a positive declination value for Hamburg and a negative for New York.

Please specify the direction in the documentation.

ayrton04, 2017-09-24:
Sorry for the delay. Yes, I will clarify this.

Sidk97, 2022-05-09:
Hi,

Is this still valid ?
I am working with navsat_transform and had some wierd behaviour, is this something that we need to consider while passing the value to magnetic_declination_radians parameter ?