cra-ros-pkg/robot_localization #380: RGB_D SLAM and IMU

yalan2017, 2017-10-13:
Hi
I am using [Rtabmap ](wiki.ros.org/rtabmap)and IMU in order to fuse my map with robot localization package but i have error and i try different way to solve but i could solve my problem. this is my [launch ](https://github.com/introlab/rtabmap_ros/blob/master/launch/tests/sensor_fusion.launch)file .
and we discussed before here [ 1 ](http://official-rtab-map-forum.67519.x6.nabble.com/tinkerforge-imu-td1886.html#a3690) [2!](http://official-rtab-map-forum.67519.x6.nabble.com/tinkerforge-imu-td1886i20.html) .
 this is my topic list 
[image]

my  result:
[image]
my frames:
[image]

Thank you 

 @jspricke 
@efernandez 
@mitchellwills

ayrton04, 2017-10-13:
Hi! This board is for reporting bugs. Please direct questions to answers.ros.org. If you have already asked a question there, can you please direct me to it?

closed 2017-10-13 by ayrton04

yalan2017, 2017-10-15:
hi
im sorry for that, i have written my problem here :  
https://answers.ros.org/question/261724/robot_localization-imu-and-visual-odometry-from-rtabmap/?answer=273047#post-id-273047
@ayrton04 
thank you