cra-ros-pkg/robot_localization #382: Backport to Indigo

jacobperron, 2017-10-20:
Due to #353, `indigo-devel` has not been able to take advantage of updates.

By changing all instances of `#include <xmlrpcpp/XmlRpcException.h>` to `#include </XmlRpcException.h>` it seems to have resolved build issues under Indigo.

I'm not sure what best practice is for backporting changes. I have the fix in [my fork](https://github.com/jacobperron/robot_localization/tree/indigo-backport), should I open a PR to `indigo-devel`? Or is there a more elegant solution?

chekakta, 2017-10-21:
@jacobperron  sorry for this question but i don't understand why when we use#include <xmlrpcpp/XmlRpcException.h>  the compilation failed
but when we use #include </XmlRpcException.h> the package is compiled with no error

jacobperron, 2017-10-21:
@chekakta No worries. I should have mentioned why.

It looks like the headers were pushed into a directory in [ros_comm](https://github.com/ros/ros_comm/commit/a8c5f9e95e6823417b67ee0da2fa0a91acd97ef5) starting with version 1.12.7. But, current ros_comm version for Indigo is 1.11.21 and so it doesn't have this change.

I posted about it in response to a similar issue on [ROS Answers](https://answers.ros.org/question/258860/about-xmlrpcppxmlrpcexceptionh/?comment=273713#post-id-273713).

jacobperron, 2017-10-21:
In retrospect, that change in ros_comm exports the old include directory for compatibility. I just confirmed that that both Kinetic and Indigo can build with the proposed change:

`#include <xmlrpcpp/XmlRpcException.h>` -> `#include <XmlRpcException.h>`

ayrton04, 2017-10-24:
Repeating my comments from the PR:

The lack of backports for new features to Indigo was not solely a result of the xmlrpcpp issue, but became an intentional move to do a better job of "freezing" releases for each ROS distribution. There are a lot of Indigo users out there, and I'm concerned about breaking functionality for them. Even though all of my tests pass, some of the changes between Indigo and Kinetic could conceivably change behavior in some subtle way that I had not anticipated.

I am still taking changes for Kinetic, though, and I put them in Lunar as well. However, some of the major changes (like nodelet support) are for Lunar only.

This obviously doesn't preclude users from building the Kinetic and Lunar branches from source.

jacobperron, 2017-10-24:
Okay. Correct me if I'm wrong, but from what I understand there will be no more feature updates to Indigo releases up to EOL (2019). And if I desire to take advantage of new features or patches I should build Kinetic or Lunar branches from source (or migrate to one of those distributions).

The only hiccup I foresee is that future changes to the Kinetic and Lunar branches may break compatibility with Indigo (similar to the one that brought up this issue), but I understand that this backwards compatibility is not a primary concern across distributions.

ayrton04, 2017-11-01:
Correct. In general, this is pretty common for ROS packages, where we add a bunch of features for new distributions and then just maintain those packages (read: fix bugs) once the distribution is released. Historically, I haven't really followed this paradigm, and tend to backport more or less everything.

However, there are a lot of people using the Indigo version, some of which are companies, and I'm loath to risk breaking something for them inadvertently. My unit test coverage isn't as broad as it ought to be, and rather than try to anticipate every possible issue, I've decided to start following the prescribed model a little better.

Having said that, I will still add minor features to Kinetic, and major (read: compatibility-breaking) changes to Lunar, hence my recommendation to use those.

jacobperron, 2017-11-01:
@ayrton04 Thanks!

closed 2017-11-01 by jacobperron