cra-ros-pkg/robot_localization #384: Frequent set pose calls stops the execution of imu callbacks

arihantb2, 2017-10-31:
I am running the ekf_localization_node with,
2 pose inputs
1 odometry input
1 imu input

When I send frequent (~1hz) set pose calls on the set_pose topic, the ekf stops processing imu input messages (stops processing imu callback) after a seemingly random instance of set pose call. 

I find that commenting out the line
```
ros::getGlobalCallbackQueue()->clear();
```
in setPoseCallback(...) seems to fix the issue. Let me know if you need any more details. I have debug log outputs from the occurrence of this issue.

ayrton04, 2017-10-31:
Interesting. That function call will exit any active callbacks and will obviously not process any more for that particular spin cycle, but the callback queue should still work afterwards. It seems odd to me that it only happens for your IMU data and nothing else. Do you have `nodelay` turned on for any of the inputs? Is your IMU data especially high-frequency?

arihantb2, 2017-10-31:
I do not have nodelay on for any of the inputs. The IMU data is coming in at about ~220hz. Although, it was easy enough to reproduce the bug at ~30hz too. What is strange is, all the other callbacks execute as expected after the event of the bug except for the imu callback. I tried adding multiple (duplicate imu configs in the params) and the callback connected to all the imu configs stop executing.

arihantb2, 2017-11-07:
Do you have any insight on what could be causing this? I can give you more information per your need.

arihantb2, 2017-11-07:
Like I mentioned earlier, commenting out the `ros::getGlobalCallbackQueue()->clear()` resolves the issue I'm seeing. Will doing that affect the node in any other way or is it harmless?

ayrton04, 2017-11-08:
So the reason I clear the callback queue is so that we don't get into awkward situations like this:

Time `t1`: Node A sends a `set_pose` request to the EKF with timestamp `t1`.
Time `t2`: Node B sends pose data time timestamp `t2` (> `t1`) to the EKF that would conflict with the data sent in the `set_pose` call.
Time `t3`: EKF receives both messages and processes them. The `set_pose` request is executed, and the EKF uses its [existing timing logic](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/ros_filter.cpp#L1589) to prevent us from processing data with any time stamps earlier than `t1`. Unfortunately, that doesn't apply to the pose data from node B, so it gets processed, and moves the robot right back where it was (or nearly there, depending on the covariance).

I have seen this happen, and while we could make the onus on the user to avoid these situations, I like the idea of getting a `set_pose` request and ignoring everything else for that spin cycle. An (admittedly awkward) alternative would be to add an `ignoreMessagesForRestOfSpin` variable, and add that as a second condition to the existing timing logic linked above.

arihantb2, 2017-11-08:
Thank you for your insight. I will keep this issue open while I look into the cause.

ayrton04, 2018-12-19:
I apparently neglected the fact that we already check for the timestamp when we received a set_pose, and don't process any messages that arrive with timestamps older than that. I removed the callback queue clearing a while ago, so this should be fine now. Clearing the global callback queue clearly has some strange side effects.

closed 2018-12-19 by ayrton04