cra-ros-pkg/robot_localization #388: Sometime state variance would be negative

lfmaster, 2017-12-06:
in the precoess of using the package, I found some strange behavior。
when looking at the log file, I found that state variance was negative。
like this :
`
------ RosFilter::integrateMeasurements ------

Integration time is 1512480946.2259142399
2 measurements in queue.
Received a measurement that was 0.10871171951293945312 seconds in the past. Reverting filter state and measurement queue...
----- RosFilter::revertTo -----

Requested time was 1512480946.078097105
Reverted to state with time 1512480946.0666921139
Restored 6 to measurement queue.

----- /RosFilter::revertTo
------ FilterBase::processMeasurement (pose0) ------
Filter is already initialized. Carrying out predict/correct loop...
Measurement time is 1512480946.078097105, last measurement time is 1512480946.0666921139, delta is 0.01140499114990234375
---------------------- Ekf::predict ----------------------
delta is 0.01140499114990234375
state is [0.53838     -3.4608     0           0           0           1.2858      -0.0062198  0.0086691   0           0           0           1.1235      -0.0065657  -0.011157   0           ]

Transfer function is:
[1           0           0           0           0           0           0.003206    -0.010945   0           0           0           0           1.8282e-05  -6.2414e-05 0           
 0           1           0           0           0           0           0.010945    0.003206    0           0           0           0           6.2414e-05  1.8282e-05  0           
 0           0           1           0           0           0           -0          0           0.011405    0           0           0           -0          0           6.5037e-05  
 0           0           0           1           0           0           0           0           0           0.003206    -0.010945   0           0           0           0           
 0           0           0           0           1           0           0           0           0           0.010945    0.003206    0           0           0           0           
 0           0           0           0           0           1           0           0           0           -0          0           0.011405    0           0           0           
 0           0           0           0           0           0           1           0           0           0           0           0           0.011405    0           0           
 0           0           0           0           0           0           0           1           0           0           0           0           0           0.011405    0           
 0           0           0           0           0           0           0           0           1           0           0           0           0           0           0.011405    
 0           0           0           0           0           0           0           0           0           1           0           0           0           0           0           
 0           0           0           0           0           0           0           0           0           0           1           0           0           0           0           
 0           0           0           0           0           0           0           0           0           0           0           1           0           0           0           
 0           0           0           0           0           0           0           0           0           0           0           0           1           0           0           
 0           0           0           0           0           0           0           0           0           0           0           0           0           1           0           
 0           0           0           0           0           0           0           0           0           0           0           0           0           0           1           ]

Transfer function Jacobian is:
[1           0           0           0           0           4.0897e-05  0.003206    -0.010945   0           0           0           0           1.8282e-05  -6.2414e-05 0           
 0           1           0           0           0           -0.00011425 0.010945    0.003206    0           0           0           0           6.2414e-05  1.8282e-05  0           
 0           0           1           9.8146e-05  7.1364e-05  0           -0          0           0.011405    0           0           0           -0          0           6.5037e-05  
 0           0           0           1.0123      0.0036018   0           0           0           0           0.003206    -0.010945   0           0           0           0           
 0           0           0           -0.0036018  1.0123      0           0           0           0           0.010945    0.003206    0           0           0           0           
 0           0           0           0           -0          1           0           0           0           -0          0           0.011405    0           0           0           
 0           0           0           0           0           0           1           0           0           0           0           0           0.011405    0           0           
 0           0           0           0           0           0           0           1           0           0           0           0           0           0.011405    0           
 0           0           0           0           0           0           0           0           1           0           0           0           0           0           0.011405    
 0           0           0           0           0           0           0           0           0           1           0           0           0           0           0           
 0           0           0           0           0           0           0           0           0           0           1           0           0           0           0           
 0           0           0           0           0           0           0           0           0           0           0           1           0           0           0           
 0           0           0           0           0           0           0           0           0           0           0           0           1           0           0           
 0           0           0           0           0           0           0           0           0           0           0           0           0           1           0           
 0           0           0           0           0           0           0           0           0           0           0           0           0           0           1           ]

Process noise covariance is:
[0.0005      0           0           0           0           0           0           0           0           0           0           0           0           0           0           
 0           0.0005      0           0           0           0           0           0           0           0           0           0           0           0           0           
 0           0           0.06        0           0           0           0           0           0           0           0           0           0           0           0           
 0           0           0           0.03        0           0           0           0           0           0           0           0           0           0           0           
 0           0           0           0           0.03        0           0           0           0           0           0           0           0           0           0           
 0           0           0           0           0           0.0007      0           0           0           0           0           0           0           0           0           
 0           0           0           0           0           0           0.001       0           0           0           0           0           0           0           0           
 0           0           0           0           0           0           0           5e-06       0           0           0           0           0           0           0           
 0           0           0           0           0           0           0           0           0.04        0           0           0           0           0           0           
 0           0           0           0           0           0           0           0           0           0.05        0           0           0           0           0           
[... cut ...]
 0           0           5.4337e-16  -1.57e-22   -1.1331e-22 0           0           0           1.4334e-13  3.0407e-27  -2.3635e-27 0           0           0           9.9422e-07  ]

---------------------- /Ekf::correct ----------------------
------ /FilterBase::processMeasurement (pose0) ------`

I dont understand how the state variance would be negative.

PS:
I use this package to fusion odom & amcl.
odom provide velocity & amcl provide pose.
when the robot velocity is near zero,the velocity variance i setting is really small ,perhaps small than 1e-10

Is there any problem with my variance setting?

or there has  a bug?

ayrton04, 2017-12-06:
In my experience, this is always the result of at least one of your input measurements having a singular covariance matrix. Please post sample input messages from all sensors.

ayrton04, 2018-02-19:
I'm not convinced that it's just `amcl`'s pose data that's giving you trouble. I'd have to see your full configuration/launch files, as well as a sample input from every sensor message. If you would like to continue this discussion, though, I'd request that you post on answers.ros.org.

closed 2018-02-19 by ayrton04