cra-ros-pkg/robot_localization #389: NavSatTransform: assumed static transform from imu frame to base frame

tcassell94, 2017-12-06:
Hi all.

There seems to be an assumed static transform from the IMU frame to the base frame. Therefore, if your IMU moves relative to the base frame (as ours does) then the output of NavSatTransform gets thrown way off. Say we drive 10m from the origin of the map frame, and then rotate our machine at our pivot point (base still, IMU rotating) NavSatTransform's estimate of our position rotates around the origin at the 10m radius. If we do not rotate at our pivot point, but instead rotate our base on the spot, the position estimate works.

I'll look into the required changes to remove the static transform (maybe just not shutting down the imu message subscriber?), but if @ayrton04 or anyone else has a quick fix, please let me know.

Cheers, Tim

ayrton04, 2017-12-06:
To which output are you referring? The `/odometry/gps` data (GPS coordinates, transformed to your robot's world frame) or `/gps/filtered` (your robot's world coordinates, transformed back into GPS coordinates)?

Does the GPS unit itself move w.r.t. the base? For `/odometry/gps`, the IMU is only needed to compute the initial transform from your robot's coordinate frame to the UTM coordinate frame. After that, [all we're doing is taking the GPS coordinates that we're fed and transforming them into UTM, and then into your robot's world frame](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/navsat_transform.cpp#L644). It's not clear to me at the moment how your robot's moving IMU would change anything for this case, since, after the transform is established, we never worry about your robot's world-frame pose or IMU data again. We just need to know the GPS coordinates.

Going the other way may present trouble, though. Can you clarify?

tcassell94, 2017-12-06:
I am referring to the `/odometry/gps` output. However, I could be wrong that it is the static transform that is the cause.

> Does the GPS unit itself move w.r.t. the base?

Yes. The GPS antenna is mounted above the IMU, both of which are on the axis of rotation that can rotate w.r.t. the base. Would a moving GPS antenna be the cause, rather than the static IMU transform?

tcassell94, 2017-12-07:
I was looking through the source code on Github and it all made sense - couldn't figure out why the rotation was occurring. But I then thought to check my local version and it turns out that I had an older version of robot_localization which was missing [this line](https://github.com/cra-ros-pkg/robot_localization/blob/cf0aa7ee634df7f6eaf6f79c3a7c81131b6d0bf5/src/navsat_transform.cpp#L435). Setting the rotation to the identity has now fixed the issue. Thanks for your quick response, feel free to close issue :smile:

ayrton04, 2017-12-07:
Someone literally just PR-ed that change, so you'd be forgiven for missing it! :) Thanks for the issue report.

closed 2017-12-07 by ayrton04