cra-ros-pkg/robot_localization #392: Invalid generated cmake config files

tfoote, 2017-12-21:
The latest release on Kinetic is generating invalid CMake Config files.

Refiling: https://github.com/clearpathrobotics/cpr_multimaster_tools/issues/2

The first failed build was triggered by the release of robot_localization: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__message_relay__ubuntu_xenial_amd64__binary/30/
[image]

ayrton04, 2017-12-21:
Whoops. Not sure what's up there. Will investigate.

ayrton04, 2017-12-21:
Not entirely sure what happened, but I tried re-blooming, and bloom wasn't happy when applying a patch to rules.em. I've got it back to a happy state, so we'll see how it behaves downstream. If it still gives us trouble, I'll roll it back and deal with it later.

tfoote, 2017-12-22:
Thanks for the quick turn @ayrton04 !
Cross reference: https://github.com/ros/rosdistro/pull/16636

ayrton04, 2017-12-22:
Not convinced it helped. Rolling back to 2.4.0.

tfoote, 2017-12-22:
Yeah, it looks like the rebuild failed: http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__message_relay__ubuntu_xenial_amd64__binary/36/console

ayrton04, 2017-12-22:
Yeah, I'm rolling it back: https://github.com/ros/rosdistro/pull/16638

Odd that it has a bunch of downstream projects, but only a [couple of them](http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__robot_localization__ubuntu_xenial_amd64__binary/) are failing.

If the rollback still fails, I'll pick it back up in the AM. Sorry for the trouble!

kenautics, 2017-12-22:
Rollback fixed it. I’ll investigate what broke and release again after the kinetic sync. Thanks for pointing it out!

ayrton04, 2017-12-22:
Rollback fixed it. I’ll investigate what broke and release again after the kinetic sync. Thanks for pointing it out!

closed 2017-12-22 by ayrton04

ayrton04, 2018-01-03:
Update: I ran a prerelease on this with a dependency level of 1, and was able to re-create the issue. We had a PR submission for some new libraries and a new node, but it wasn't installing said objects. I've fixed it and am running a prerelease test again. If that succeeds, I'll bloom.

Sorry for the trouble!