cra-ros-pkg/robot_localization #396: Sensor Fusion with robot_localization

eroniki, 2018-01-25:
Hello,

I am wondering how robot_localization package is formulating the sensor fusion problem if more than one sensor is used during the localization. Specifically, the sensors operate at different frequencies how this package handle the fusion of the measurements coming from different sensors.

Considering a robot equipped with an IMU, working at 50 Hz, and a camera providing visual odometry at 20 Hz, how this package is employing the measurements as the package localizes.

Thanks in advance.

Best regards,
Murat

ayrton04, 2018-01-25:
Hi,

Would you mind posting questions on answers.ros.org and pointing me at them? This space is mostly for bugs and feature requests.

Thanks!

closed 2018-01-25 by ayrton04

eroniki, 2018-01-25:
Hey, Tom
I posted the original question on the ROS answers. The link is provided below: 

https://answers.ros.org/question/280831/specifics-of-sensor-fusion-algorithm-of-robot_localization-package/ 

Thanks in advance!