cra-ros-pkg/robot_localization #397: Publishing frequency of navsat_transform when publish_filtered_gps is set is lower than expected

nventuro, 2018-02-08:
`prepareFilteredGps` only returns `false` (and causes `run` not to publish) when `odom_updated_` is set, which happens every time `odomCallback` is called. This syncing mechanism using flags fails when the `frequency` parameter of `navsat_transform_node` is similar to the `/odometry/filtered` publishing rate (and relatively high, e.g. 100Hz).

On my setup, [the queue length of 1 on the subscriber](https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/navsat_transform.cpp#L162) causes it to lose around 70% of the published messages, reducing the `/gps/filtered` publishing rate to 30Hz. Increasing the queue length doesn't solve this, since the messages are processed in a bursty fashion by the callback, causing older (and unprocessed) messages to the overwritten by the latest one. 

I believe this could be solved by storing the `/odometry/filtered` messages in a properly sized queue, and processed in a FIFO order.

ayrton04, 2018-02-19:
Can you post your config/launch file and perhaps a bag? I'd like to be able to reproduce before I comment. What happens if you increase the `navsat_transform_node` to 200Hz? I'd also try adding the `tcpnodelay` flag to the subscriber to see if that helps. Are you seeing this when playing back data in a bag file, or live?

closed 2018-04-18 by nventuro

Farid010, 2019-05-28:
@nventuro I have the same problem. Did you find any solution?

YairBarzilayBWR, 2023-11-06:
Why was this closed? i have the same issue.