cra-ros-pkg/robot_localization #399: navsat_transform should wait for valid time

agoluberg, 2018-03-02:
Filter node is waiting for the valid time (`ros::Time::waitForValid()`). The navsat_transform should use the same mechanism. When using the node in simulator in paused mode, the delay parameter for example is used up before I can start the simulator and therefore there is no actual delay compared to other nodes.

ayrton04, 2018-03-02:
Sounds reasonable! Care to submit a PR?

ayrton04, 2018-06-19:
Would you happen to have time to test [this branch](https://github.com/locusrobotics/robot_localization/tree/399-ntn-valid-time)?

agoluberg, 2018-06-19:
Sorry, I actually created the same modification but didn't submit it here. We are using the exact same code and it seems to work just fine. I don't have a possibility to test exactly this branch. We start Gazebo in pause mode and the node now waits until unpause is initiated.

ayrton04, 2018-06-19:
Did you [also remove the ros::Time::init()](https://github.com/cra-ros-pkg/robot_localization/compare/cra-ros-pkg:f0b540b...locusrobotics:de60fdb#diff-dc270f64cc0f5f1b532a944e353b01e0L79)?

agoluberg, 2018-06-19:
Yes. But if I think about that, I don't remember why exactly. I checked that EKF nodes both use the same code.  With init, the `g_use_sim_time` will be set to false. But that caused the problems (if I remember correctly). This is a problem only if used with simulated time. Live systems work well without this `waitForValid()` call.

agoluberg, 2018-06-19:
Yes, of course, init is done in the node.cpp. And yes, it is working as expected. Thanks!