cra-ros-pkg/robot_localization #400: Fusion of multiple absolute odometry sources

tik0, 2018-03-05:
Dear folks, I try to integrate multiple absolute position trackings with the robot_localization.
Unfortunately, the EKF jumps back and forth between odom0 and odom1 localizations (they originate from external tracking systems like GPS).
Is there a setup or launch file configuration which fuses both estimates in an optimal fasion: like weighted mean?

Here is my launch file configuration:
```
<?xml version="1.0"?>
<launch>

  <arg name="id" default="0"/> 

  <node pkg="robot_localization" type="ekf_localization_node" name="ekf" output="screen" > 

    <!-- Frequency of the main run loop. -->
    <param name="frequency" value="30"/>
    <param name="sensor_timeout" value="1.0"/>
    <param name="map_frame" value="map"/>
    <param name="odom_frame" value="world"/>
    <param name="base_link_frame" value="base_link/base_link"/>
    <param name="world_frame" value="world"/>
    <param name="two_d_mode" value="true"/>

    <remap from="odometry/filtered" to="odometry/filtered"/>

    <param name="odom0" value="/external_tracker_1/cam/odom"/>
    <rosparam param="odom0_config">[true,  true,  true, 
                                    true, true, true, 
                                    false, false, false, 
                                    false, false, false,
                                    false, false, false]</rosparam>
    <param name="odom0_differential" value="false"/>
    <param name="odom0_relative" value="false"/>

    <param name="odom1" value="/external_tracker_2/cam/odom"/>
    <rosparam param="odom1_config">[true,  true,  true, 
                                    true, true, true, 
                                    false, false, false, 
                                    false, false, false,
                                    false, false, false]</rosparam>
    <param name="odom1_differential" value="false"/>
    <param name="odom1_relative" value="false"/>

    <!-- More to come -->
  </node>

</launch>
```

ayrton04, 2018-03-05:
The EKF *is* fusing the estimates in an optimal fashion, as a weighted mean. The Kalman gain computes the optimal weighting for the state and measurement based on their relative covariances, and then fuses the measurement.

If your state estimate is jumping back and forth between your two measurement sources, it means that their covariances are probably too low. If sensor 1 reads some value with variance of 0.1 meters, and sensor 2 reads a value that is 5 meters away with a variance of 0.1 meters, then at least one of your sensors is lying. You might also want to drop your `process_noise_covariance` for your pose variables. You didn't post it, so I'm not sure what it is.

You should also make sure that your two pose estimates are being reported in the same frame.

You can also take your less accurate input and turn on `differential` mode for it.

ayrton04, 2018-03-05:
(Also, if you have further questions, I'd ask that you please post them on answers.ros.org).

tik0, 2018-03-05:
Thanks for that hint.
A first try seems to solve the issue, but I need more investigation on that.
I will ask further implementation questions on answers.ros.org!

ayrton04, 2018-03-19:
Great!

closed 2018-03-19 by ayrton04