cra-ros-pkg/robot_localization #401: Question regarding sensor_timeout.

aalapshah12297, 2018-03-08:
Hello, I am using robot_localization on a ground robot for getting filtered odometry by combining IMU data and raw wheel odometry. The problem is that my sensor frequencies are ~10 Hz for both.But I want to have odometry at around 20 Hz in order for my control to work accurately. So I have set the frequency to 20 Hz in the ukf parameters. This implies a sensor timeout of 0.05 (it is both the default value, and the value that I have passed to it via the .yaml file). Thus, robot_localization will be running some prediction-only cycles between prediction-measurement cycles.
The odometry that I get this way is good, but there is a problem - the difference between the timestamps of consecutive messages is very erratic now (it was not so when I ran robot_localization at 10 Hz). Sometimes the timestamps are as close as 0.0001s. This causes problems in a particular SLAM node that I am using.
My question is: Is this not a recommended way of using robot_localization? Is the sensor_timeout field not supposed to be used for predicting intermediate odometry between actual measurements (and only to be used for long durations where the sensor actually stops working)?

ayrton04, 2018-03-08:
Thanks for the question. However, we ask that all questions be directed to http://answers.ros.org.

closed 2018-03-08 by ayrton04

aalapshah12297, 2018-03-13:
Nevermind. I looked into the source and found that robot_localization uses a parameter called `predict_to_current_time`. Setting this to `true` via the parameter file fixed my problem. The timestamps are uniform now.