cra-ros-pkg/robot_localization #402: Add additional state

virtualclone, 2018-03-14:
I would like to ask:
Can I add additional state to the 15-dimensional state ?
For example, I want to estimate the force by EKF or UKF.
Thanks for your reply!

ayrton04, 2018-03-19:
That won't be a trivial change. Supporting state vectors with different variable sets would require a fairly massive rewrite. You'd have to change the state transition functions, calculate the Jacobians, etc. If we were going that way, I'd suggest a plugin-based architecture, but that's going to require a fairly large amount of effort.

virtualclone, 2018-03-30:
And another question, the published topics are `odometry/filtered`(the message type is `nav_msgs/Odometry`) and `/accel/filtered`(the message type is `geometry_msgs/AccelWithCovarianceStamped`) . Does it support other message type such like `geometry_msgs/PoseWithCovarianceStamped`? Because I would like to estimate angular velocity or force. Or I need to rewrite the code. Thanks!

ayrton04, 2018-03-30:
No, it doesn't support that output, but note that a `nav_msgs::Odometry` _contains_ a `geometry_msgs::PoseWithCovariance` (see the definition [here](http://docs.ros.org/api/nav_msgs/html/msg/Odometry.html)), and it has a header, so it has a stamp. In other words, it contains the same data as a `PoseWithCovarianceStamped`, as well as velocity data.

However, the filter doesn't have force as a state variable. It estimates position, orientation, linear velocity, angular velocity, and linear acceleration.

I'd recommend giving the wiki a thorough read through to get a sense of what the filter is estimating.

closed 2018-03-30 by ayrton04