cra-ros-pkg/robot_localization #403: Issue : Program navsat_transform_node does not subcribe to IMU topic

antoninhaulot, 2018-03-29:
Hello,

I am currently trying to use the navsat_transform_node in a ROS-Gazebo simulation. I'm using Ubuntu 16.04 LTS, with ROS kinetic and Gazebo 7.

My issue is that the navsat_transform_node does not subscribe to the IMU topic, even if the type of the IMU topic published by gazebo is sensor_msgs/Imu. 
I used the navsat launch file from this tutorial : [http://docs.ros.org/api/robot_localization/html/integrating_gps.html](url)
The IMU's plugin used by gazebo is the "libgazebo_ros_imu.so". I do not understand why the navsat is not subscribing to the IMU, despite it's well subscribed to the odom and gps topic. 

Do you have any ideas why the navsat node is not subscribing to the IMU topic ?

ayrton04, 2018-03-29:
Hi, would you mind asking this on answers.ros.org and pointing me to it? Please post your full config, and a sample message from every sensor input. Also make sure you give me the topics names, please.

closed 2018-03-29 by ayrton04

antoninhaulot, 2018-04-04:
Here is the question on answers.ros.org : https://answers.ros.org/question/287090/navsat_transform_node-does-not-subscribe-to-imu-topic/