cra-ros-pkg/robot_localization #407: Potential issue with UKF interpreting pitch and roll as yaw

msy22, 2018-05-31:
# Problem Description (TL;DR)

I've noticed lately that when the robot is travelling on a slope, the odometry output from the global ukf_localization_node develops a slight heading error, which makes the robot look like it is "skating" sideways slightly. E.g:

[image]

So in the above image, even though the robot is heading in a straight line from left to right, it appears to be yawing slightly. This doesn't happen at all if I drive the robot around on a perfectly flat surface, such as in concrete carpark or on a flat field. Specifically, I'm looking at the odometry output of a global UKF on the topic `/odometry/filtered/global`. You can see this skating motion in the first three bag files I have provided below.

What seems to be happening is that **when the robot pitches or rolls, the UKF interprets some of that motion as yawing.**

Unfortunately the issue seems to originate within either the `navsat_tranform` or global `ukf_localization_node`. At this stage, I believe this is a bug but it may be that that my data is somehow fatally flawed or that this is a limitation of the UKF engine. Regardless, I thought that it would be best to log an issue here, and I'd appreciate any assistance in resolving this problem.

# Problem Description (detailed)

I initially thought that this was because slip between the wheels and ground when traveling on a slope was creating a yawing motion. However, no matter how much I played with the configuration of the UKF nodes, I wasn't able to get rid of this skating motion. 

I then conducted a "tilt test" - essentially carefully pitching the robot forward, rolling it right, pitching it back and then rolling it left. This is so the wheel odometry remains zero and the robot experiences carefully controlled angular acceleration in one direction only. In this case, the wheel encoder data (odometry on the `/jackal_velocity_controller/odom` topic) reports no movement and so (in theory) cannot affect the orientation of the UKF output, and yet I see UKF odometry output that looks like this:   

[image]

Where very clearly the odometry output has yawed, even though the robot itself never physically did. And when inspecting the odometry outputs:

[image]

you can see where the wheel odometry (`/jackal.../odom`, black) shows the robot moved forward, and then stopped moving. So this test rules out the wheel encoder odometry as the source of the error. In addition, if I inspect the raw IMU roll, pitch and yaw values, I see:

[image]

Which clearly shows a negligible amount of yaw being reported by the IMU, and correctly shows the roll and pitch angular acceleration experienced by the robot. So from this data I can only assume that the problem is originating from with `navsat_transform` or one of the `ukf_localization_node`s, I suspect the global UKF. I've provided some bagfiles below that demonstrate the issue, as well as the bag file I used to collect all of this data. 

I've also noticed that the raw GPS data shows the correct motion in the UTM grid:

[image]

But the odometry already shown follows a curved trajectory that doesn't seem to move uniformly in the same linear 'X' pattern shown in the UTM coordinates.

# Example Bagfiles

Bagfile demonstrating skating when traveling perpendicular to direction of slope: [slope_test1](https://drive.google.com/open?id=1aS-PAZRy6j23ztb0T3h993YVyNvuHAo9)

Bagfile demonstrating skating when traveling parallel to direction of slope: [slope_test2](https://drive.google.com/open?id=1qCLJLoqhh3yV_oiOCbch-SuKc9UVOHmx)

Bagfile demonstrating yaw offset when robot begins on a slope rather than flat ground: [slope_test3](https://drive.google.com/open?id=1qc7OZ6hpxUDpfkiaHgcc_J_upLinTH0Z)

Bagfile with forward motion and then careful pitch forward, roll right, pitch back, roll left motion: [tilt_test](https://drive.google.com/open?id=1bH9T56k7BFRIWCyLifdyMYPurFhwpv_3) (note that this bagfile contains a short period of forward motion before the robot is pitched and rolled).

# System details

I'm running a "standard" r_l setup with two instances of ukf_localization_node (one local, one global) and one instance of navsat_transform. I'm fusing IMU, wheel encoder and GPS odometry into the global UKF and just IMU/encoder data into the local UKF, see the launch details for more details. 

- **Robot:** Clearpath Jackal (real, not simulated)
- **OS:** Ubuntu 14.04 desktop (laptop), Ubuntu 14.04 server (robot)
- **r_l version:** 2.3.4
- **Launch file:** [r_l_launch_file](https://drive.google.com/open?id=1YJj0CNmPYu_GsX3LZ3tA_qAtiFjuEKRB)
- **Odometry topics:** 
  - `/jackal_velocity_controller/odom`: raw wheel encoder odometry, black
  - `/odometry/filtered/local`: encoder and IMU data fused, blue 
  - `/odometry/filtered/global`: encoder, IMU and GPS odometry fused, red 
  - `/odometry/gps` : GPS odometry, green

ayrton04, 2018-05-31:
Thanks for the very detailed report! Nice work. I'll review and digest as time allows.

ayrton04, 2018-05-31:
BTW, have you tried this with the EKF?

msy22, 2018-06-04:
No I haven't actually. I'll strip the odometry from those bagfiles and run `ekf_localization_node`s over them with the same configuration to see if the output also looks like it's skating.

msy22, 2018-06-05:
Unfortunately it appears the same thing happens with the `ekf_localization_node`. For example, here is the UKF odometry displayed in RViz:

[image]

And here is the EKF odometry:

[image]

In addition, the tilt test results are arguably worse, UKF result:

[image]

EKF result:

[image]

In fact it looks like the EKF isn't good at interpreting the rolling motion, and thinks the GPS odometry is actually strafing the robot to it's side.

However, does the fact that this error appears with both the EKF and UKF indicate that it is an issue with any shared code they might have? Or with the input data I'm using?

msy22, 2018-06-05:
Oh, and here are the bagfiles using the same input data but with the EKF instead of UKF

- [slope_test1_ekf.bag](https://drive.google.com/open?id=1aJuTfY_o-hSGx5SGEzBXCX6-BHyzOQ02)
- [tilt_test_ekf.bag](https://drive.google.com/open?id=1b4iwPA4Upwn7gfSnFCZQ1CvUOYEbxb6p)

For reference, I used the exact same launch file, but launching EKF nodes instead of UKF nodes, and without the UKF tuning parameters: [r_l_launch_file_ekf.launch](https://drive.google.com/open?id=1qn2mGQwI21Ubg18sgwnsJ_vm9PpkmpiR)

ayrton04, 2018-06-05:
No, it's good that it's not in the filters themselves, but in a data preparation step. Will look into it when I have some cycles, thanks.

msy22, 2018-06-06:
No problem, as always I appreciate your help whenever you have the time to provide it. I see this behaviour pretty consistently on grassy, sloped terrain so if the problem is caused by the data, then I assume it is related to my configuration parameters for `navsat_transform` and `ukf_localization_node`.

Also, for reference I have a few extra IMU topics in these bag files, the only one I'm using in this situation is `/imu/filtered/debiased` which is essentially data copy-pasted from the raw data in the `/imu/data` topic, but with a bias subtracted that a node I've written automatically calculates whenever the robot is stationary. The bias is published on the `/imu/filtered/bias` topic.

msy22, 2018-06-06:
Unfortunately the bug in my zero velocity update wasn't the problem, the data doesn't look any different from the original images. I have revised the bagfiles though, here are the new ones:

- [slope_test1_corrected.bag](https://drive.google.com/open?id=1X5D1_sKX5f1S9wbRqKnCnNqgP9f8f-hs)
- [slope_test2_corrected.bag](https://drive.google.com/open?id=1tWM1gtykzAmG_jArf-UJ_FGEjpUQ7ror)
- [slope_test3_corrected.bag](https://drive.google.com/open?id=1ijQB0IJYFjIroDnx2fccfqaKDp2wr4ru)
- [tilt_test_corrected.bag](https://drive.google.com/open?id=1BAkOqI9Jukv1pu8FdPxBnU3XancBSCpp)

ayrton04, 2018-06-11:
Looking at the images you posted, and taking a quick look at the bags, it looks to me like your IMU raw heading does not match your GPS. In other words, if you drive your robot in a long, straight line, the `atan` of the end points should be ~ what your IMU yaw reads. I'll verify and get back to you.

ayrton04, 2018-06-14:
I haven't forgotten this, BTW. I need to work through your tilt tests and make sure that's just another symptom of the same issue.

msy22, 2018-06-17:
No problem Tom. Regarding my IMU, it is a fairly "cheap and cheerful" model, so the heading is often off by up to about 5 degrees, some times 10 on a bad day. It also tends to drift over time, which the ZUPT (and the data it produces on the `/imu/filtered/debiased` topic) largely deals with. Unfortunately that's about as accurate as it gets, even with calibration.

ayrton04, 2018-06-19:
There are two parts to my answer. Here's part 1.

I stripped the bag file of all data aside from raw sensor input, and ran everything live. I removed `navsat_transform_node` just to isolate the issue. Here's a video of what I saw (for some reason, GitHub won't let me add the mp4 directly, so it's zipped):

[matt_rpy.mp4.zip](https://github.com/cra-ros-pkg/robot_localization/files/2114772/matt_rpy.mp4.zip)

I see the robot move forward a tad, then it gets pitched, then it seems to turn in yaw, then pitched the other way, then it seems to turn in yaw again. My first step was to plot the raw sensor data. Here is an Octave plot of the `/imu/filtered/debiased` topic, which you are feeding directly into the EKF:

[image]

Each time you roll, your filtered IMU data is also producing strong motion in yaw. I see your plots above used raw IMU, but that's not what you're feeding to the filter. It looks to me like whatever you are using to debias the filter is giving you false motion in yaw.

ayrton04, 2018-06-19:
Part 2.

You also have some issues with your IMU w.r.t  its heading, as compared to what your GPS is saying.

First, consider this output:

[image]

That is what I get when I run your setup (that's the `/odometry/filtered/global data in the map frame), and with the output without the `navsat_transform_node`. See how the arrows point exactly where the robot is facing? It makes sense, since the only absolute pose data we have in that state estimate is the absolute orientation coming from the IMU. So the velocities just get integrated and we move forward at whatever direction the IMU says we're facing.

Now, consider this output:

[image]

That's the exact same output, except I am drawing the output of `navsat_transform_node`, though I am NOT fusing it. See how they utterly disagree with each other? If `navsat_transform_node` is configured correctly, those two paths should approximately agree, or at least the relative heading between two points ought to agree. But they don't.

Looking at the `slope_3` test, I plotted the GPS locations at the start end end of the long, straight driving bit at the end. Here's what Google Earth shows:

[image]

Now, Google Earth's coordinate frame differs, so it believes true north is 0 (just talking about yaw here), and the heading increases clockwise. Your IMU need to increase counter-clockwise, so assuming your IMU also reads 0 at true north (once you take care of magnetic declination), Google Earth's heading of 154.32 degrees is -154.32 degrees for your IMU, or -2.6933921017 radians. However, during that straight driving portion, your robot's IMU was reporting a yaw of -1.975. That's a 42-degree difference.

So what's the upshot of all of this? When you are driving, you are going to get false lateral motion, as though your robot is getting dragged sideways relative to its direction of motion. This will create a sensation that your robot has a weird yaw, because your IMU says you are facing, e.g., 10 degrees, but your GPS relative motion suggests you are moving at 25 degrees.

So what you need to do is, independent of the settings for correcting the magnetic declination and the yaw offset (which we use to make 0 face east instead of north), you should make sure that your IMU heading and the heading obtained from relative GPS poses agree. If you drive straight, and your GPS says you went one direction, but your IMU says you're facing another, you're going to get the output you posted above.

To test: point your robot at true north. Drive it 10 meters. Differentiate the GPS poses and verify that you have ~0 heading. Account for magnetic declination in the IMU, and you should also see ~0 heading. If you don't, that's a problem.

msy22, 2018-06-20:
Ok, addressing the IMU RPY first, it looks like the problem is that I've only been looking at the angular velocity rather than the quaternion-derived absolute roll, pitch and yaw. I (naively) assumed that the data from the magnetometer, gyroscopes and accelerometer would be consistent, but this obviously isn't the case. I re-wrote my plotting script from scratch and this is what I get for `20180531_tilt_test_corrected_imu`:

[image]

[image]

[image]

So the actual RPY plot matches your Octave plot, which is good. Otherwise you're right, it looks like the magnetometer is reporting a yaw when it shouldn't be - while the gyroscopes correctly report no angular velocity about the Z axis, and the linear acceleration looks reasonable as well.

As for the cause, I don't think it's my ZUPT since I've changed it to only remove the steady-state bias from the angular acceleration (since those are the only values I can expect to be zero if the robot is stationary) - the ZUPT doesn't touch the orientation or linear acceleration. I've plotted both the raw and debiased data in the plots above, using different line styles so that when the raw and debiased data agree I should see only one coloured line for each axis. This I think proves that the problem exists in the raw data and therefore stems from the IMU's magnetometer.

As for correcting this issue, I last calibrated the jackal's IMU only a week or two before that tilt test, so either the calibrate sequence bugged out on that particular occasion, or is insufficient. I suspect the latter because Clearpath's automated script only rotates the robot about the horizontal axis, the instructions on their website don't mention anything about pitching or rolling the robot, and obviously it can't do that itself... so I don't see any way for the IMU to calibrate it's pitch or roll. I'll re-calibrate the IMU when I next can and repeat the tilt test to see if the problem persists. 

I'll now work on addressing the odometry...

msy22, 2018-06-27:
Apologies for taking so long to address the odometry issues. It's taken a while to re-write my plotting code and recent weather has stopped me from collecting data outside. 

Ok, so to begin with I think the heading reported by my IMU is correct. I did a few "drive in a straight line" tests like you suggested, and in each the IMU heading roughly matches the GPS heading:

Trajectory:
[image]

True heading from IMU and GPS:
[image]

In each case the test was conducted on flat ground, whereas when I look back at the respective headings during the slope tests, they deviate significantly in several places, which is what we expect since we've already established that the IMU reports a yaw when it pitches or rolls:

[image]

[image]

[image]

So I think this just confirms that the issue manifests only when the robot pitches or rolls. 

To more directly address your previous post, how did you create that second image? The one showing the outputs of `/odometry/gps` and `/odometry/filtered/global` deviating? I haven't been able to re-create it. 

I tried by cutting out the `/odometry/gps` data as an input to the global UKF instance, but this essentially just makes it a carbon copy of the local UKF instance, and when I plot the odometry for that dataset (`slope_test1`) I get a different output:

[image]

Where obviously they don't line up perfectly, but the orientation of each trajectory broadly seems to match. So unless I'm calculating my headings wrong (and I should hope not after spending the best part of a whole day bug fixing and checking everything) then I don't think there is any serious discrepancy between the IMU and GPS heading... provided the robot is generally level. I think this also means then that my configuration of `navsat_transform` is correct, and that it can be ruled out as part of the problem.

As for fixing the issue, re-calibration didn't fix it. My tilt tests largely look the same before and after recalibration. The Clearpath automated script for calibrating the compass obviously doesn't take into account the roll or pitch because I tried moving the robot during the calibration process... and it didn't work. 

Either way, there's obviously no action needed at your end, as it looks like the problem is entirely being caused by my physical IMU. I'll mess around some more on my end, and maybe trial a launch files that don't fuse yaw into the UKF, but otherwise you can probably go ahead and close this issue.

ayrton04, 2018-06-27:
Re: how I generated the plot in question, I just removed the GPS as an input to your second-tier EKF.

In any case, it looks like you have a good handle on this now. Maybe file a ticket about IMU cal for the Husky? I'll close for now, but feel free to re-open if you find compelling evidence of an issue in r_l.

closed 2018-06-27 by ayrton04