cra-ros-pkg/robot_localization #409: Fusion of poses only

tik0, 2018-06-10:
I'd like to fuse positioning measurements which send their estimates via PoseWithCovarianceStamped messages.
However, if I just use `pose*` parameters, the `robot_localization` node won't start.

Here is an example configuration:
```
  <arg name="id" default="4"/>
  <node pkg="robot_localization" type="ekf_localization_node" name="ekf_wheel_odometry_$(arg id)" output="screen" > 
    <!-- Frequency of the main run loop. -->
    <param name="frequency" value="30"/>
    <param name="sensor_timeout" value="1.0"/>
    <param name="map_frame" value="map"/>
    <param name="odom_frame" value="odom"/>
    <param name="base_link_frame" value="$(arg id)/base_link"/>
    <param name="world_frame" value="odom"/>
    <param name="two_d_mode" value="true"/>
    <param name="publish_tf" value="true"/>
    <remap from="odometry/filtered" to="/$(arg id)/odometry/filtered/"/>
    <param name="pose0" value="/pose/$(arg id)"/>
    <rosparam param="pose0_config">[true,  true,  true, true, true, true, true, true, true, true, true, true, true, true, true]</rosparam>
    <param name="pose0_differential" value="false"/>
    <param name="pose0_relative" value="false"/>
  </node>
```
With the resulting error message:
```
[FATAL] [1528560370.359265905]: Could not read sensor update configuration for topic odom0 (type: 0, expected: 7). Error was type error
```

As soon as I configure the `odom0` input and feed it, everthing works.
Is this a possible bug, or is it intended that at least one odometry pose has to exist?
If it so, then i think it is not doumented.

ayrton04, 2018-06-11:
Are you sure you don't have a dangling reference to `odom0` on your parameter server? Did you restart your ROS core before running? Do a `rosparam list | grep odom0` and see what you get.

That method that prints that error message is [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L2272). That method is called for `odom` [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L994), for `pose` [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L1118), for `twist` [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L1200), and for `imu` [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L1290). The only way that method can get called for `odom0` is if `odom0` is defined somewhere. That check is done [here](https://github.com/cra-ros-pkg/robot_localization/blob/f0b540bbd0d645c31872aabbf1ced3c2491c9dbb/src/ros_filter.cpp#L956).

ayrton04, 2018-06-11:
I made a launch file that contains only your launch configuration. Here's the output I see when I run it. Looks normal.

```
SUMMARY
========

PARAMETERS
 * /ekf_wheel_odometry_4/base_link_frame: 4/base_link
 * /ekf_wheel_odometry_4/frequency: 30
 * /ekf_wheel_odometry_4/map_frame: map
 * /ekf_wheel_odometry_4/odom_frame: odom
 * /ekf_wheel_odometry_4/pose0: /pose/4
 * /ekf_wheel_odometry_4/pose0_config: [True, True, True...
 * /ekf_wheel_odometry_4/pose0_differential: False
 * /ekf_wheel_odometry_4/pose0_relative: False
 * /ekf_wheel_odometry_4/publish_tf: True
 * /ekf_wheel_odometry_4/sensor_timeout: 1.0
 * /ekf_wheel_odometry_4/two_d_mode: True
 * /ekf_wheel_odometry_4/world_frame: odom
 * /rosdistro: hotdog
 * /rosversion: 1.14.1

NODES
  /
    ekf_wheel_odometry_4 (robot_localization/ekf_localization_node)

ROS_MASTER_URI=http://localhost:11311

process[ekf_wheel_odometry_4-1]: started with pid [29687]
[ INFO] [/ekf_wheel_odometry_4]: Waiting for valid clock time...
[ INFO] [/ekf_wheel_odometry_4]: Valid clock time received. Starting node.

```

Now, if I create a dangling reference to odom0 (`rosparam set /ekf_wheel_odometry_4/odom0 test`) and run again, I get this:

```
SUMMARY
========

PARAMETERS
 * /ekf_wheel_odometry_4/base_link_frame: 4/base_link
 * /ekf_wheel_odometry_4/frequency: 30
 * /ekf_wheel_odometry_4/map_frame: map
 * /ekf_wheel_odometry_4/odom_frame: odom
 * /ekf_wheel_odometry_4/pose0: /pose/4
 * /ekf_wheel_odometry_4/pose0_config: [True, True, True...
 * /ekf_wheel_odometry_4/pose0_differential: False
 * /ekf_wheel_odometry_4/pose0_relative: False
 * /ekf_wheel_odometry_4/publish_tf: True
 * /ekf_wheel_odometry_4/sensor_timeout: 1.0
 * /ekf_wheel_odometry_4/two_d_mode: True
 * /ekf_wheel_odometry_4/world_frame: odom
 * /rosdistro: hotdog
 * /rosversion: 1.14.1

NODES
  /
    ekf_wheel_odometry_4 (robot_localization/ekf_localization_node)

ROS_MASTER_URI=http://localhost:11311

process[ekf_wheel_odometry_4-1]: started with pid [30480]
[ INFO] [/ekf_wheel_odometry_4]: Waiting for valid clock time...
[ INFO] [/ekf_wheel_odometry_4]: Valid clock time received. Starting node.
[FATAL] [/ekf_wheel_odometry_4]: Could not read sensor update configuration for topic odom0 (type: 0, expected: 7). Error was type error

```

ayrton04, 2018-06-11:
I made a launch file that contains only your launch configuration. Here's the output I see when I run it. Looks normal.

```
SUMMARY
========

PARAMETERS
 * /ekf_wheel_odometry_4/base_link_frame: 4/base_link
 * /ekf_wheel_odometry_4/frequency: 30
 * /ekf_wheel_odometry_4/map_frame: map
 * /ekf_wheel_odometry_4/odom_frame: odom
 * /ekf_wheel_odometry_4/pose0: /pose/4
 * /ekf_wheel_odometry_4/pose0_config: [True, True, True...
 * /ekf_wheel_odometry_4/pose0_differential: False
 * /ekf_wheel_odometry_4/pose0_relative: False
 * /ekf_wheel_odometry_4/publish_tf: True
 * /ekf_wheel_odometry_4/sensor_timeout: 1.0
 * /ekf_wheel_odometry_4/two_d_mode: True
 * /ekf_wheel_odometry_4/world_frame: odom
 * /rosdistro: kinetic
 * /rosversion: 1.14.1

NODES
  /
    ekf_wheel_odometry_4 (robot_localization/ekf_localization_node)

ROS_MASTER_URI=http://localhost:11311

process[ekf_wheel_odometry_4-1]: started with pid [29687]
[ INFO] [/ekf_wheel_odometry_4]: Waiting for valid clock time...
[ INFO] [/ekf_wheel_odometry_4]: Valid clock time received. Starting node.

```

Now, if I create a dangling reference to odom0 (`rosparam set /ekf_wheel_odometry_4/odom0 test`) and run again, I get this:

```
SUMMARY
========

PARAMETERS
 * /ekf_wheel_odometry_4/base_link_frame: 4/base_link
 * /ekf_wheel_odometry_4/frequency: 30
 * /ekf_wheel_odometry_4/map_frame: map
 * /ekf_wheel_odometry_4/odom_frame: odom
 * /ekf_wheel_odometry_4/pose0: /pose/4
 * /ekf_wheel_odometry_4/pose0_config: [True, True, True...
 * /ekf_wheel_odometry_4/pose0_differential: False
 * /ekf_wheel_odometry_4/pose0_relative: False
 * /ekf_wheel_odometry_4/publish_tf: True
 * /ekf_wheel_odometry_4/sensor_timeout: 1.0
 * /ekf_wheel_odometry_4/two_d_mode: True
 * /ekf_wheel_odometry_4/world_frame: odom
 * /rosdistro: kinetic
 * /rosversion: 1.14.1

NODES
  /
    ekf_wheel_odometry_4 (robot_localization/ekf_localization_node)

ROS_MASTER_URI=http://localhost:11311

process[ekf_wheel_odometry_4-1]: started with pid [30480]
[ INFO] [/ekf_wheel_odometry_4]: Waiting for valid clock time...
[ INFO] [/ekf_wheel_odometry_4]: Valid clock time received. Starting node.
[FATAL] [/ekf_wheel_odometry_4]: Could not read sensor update configuration for topic odom0 (type: 0, expected: 7). Error was type error

```

ayrton04, 2018-06-14:
Closing. If you have further trouble, please ask on answers.ros.org.

closed 2018-06-19 by ayrton04