cra-ros-pkg/robot_localization #412: Unable to use navata_transform node

Mondo92, 2018-06-29:
Hi everybody,

I would ask for help to use the navata_transform node. I've a robot which publishes only:
- a GPS message topic (sensor_msgs/NavSatFix message) 
- a IMU message topic ( sensor_msgs/Imu )
I'm trying to simply convert the latitude/longitude/altitude in the relative x/y/z position without running any other node in the robot (rviz, ekf/ukf,...). 
Is it possibile in your opinion? Can I reach my purpose even without any odometry source? 
I've already tried to use wait_for_datum flag but the node doesn't seem to work, it doesn't print any ROSINFO on the terminal even if the node starts correctly, with all the topics correctly published.

Thanks in advance.

ayrton04, 2018-06-29:
Can you please post this question on answers.ros.org and point me to it? Thanks.

closed 2018-06-29 by ayrton04

Mondo92, 2018-06-29:
I've already post the question on answer.ros.org, thanks for your prompt reply.
I'm sorry but I can't find out how to tag users in this site. Anyway, the link of the question is below.

https://answers.ros.org/question/295803/unable-to-use-navsat_transform_node/