cra-ros-pkg/robot_localization #417: Feature request: discard erroneous measurements

rokusottervanger, 2018-09-17:
Using statistical methods, it is possible to draw conclusions about the validity of the combination of a measurement and the predicted state. If a measurement (including its covariance) is not consistent with the predicted state estimation (and its respective uncertainty), it implies that either the physical model or the measurement is incorrect. I would assume that in most cases, the measurement would be incorrect. An example of this is the occurrence of multipath effects in GPS and UWB. Using this conclusion about the validity of the measurement, one can choose to use the measurement or discard it. 

A reference of this method (I know, it's not very recent): https://www.stats.ox.ac.uk/~caron/Publications/J_Information_Fusion_2004.pdf 

I'm going to experiment with this method for a project I'm working on. I'm not saying this will be the holy grail. I already see a problem where measurements are discarded for some time, the covariance of the prediction grows, enabling faulty measurements to be fused. So it should definitely be an option that can be disabled. But if I were to implement this, would you be interested in a PR?

ayrton04, 2018-09-17:
This is the reason for the `*_rejection_threshold` parameters. They are specified as Mahalanobis distances, and they will use the EKF's current covariance estimate (along with the covariance measurement from the sensor) to control whether to reject the measurement:

https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/ekf.cpp#L185
https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/filter_base.cpp#L380

rokusottervanger, 2018-09-17:
> to determine whether to reject the measurement:
> EDIT: let me know if this doesn't solve your use case, though.

Ah that's nice. I must admit, it's been a while since I looked at the code,
so I didn't know it was in there. We'll have a look, try it, and get back
to you if it doesn't solve the case. Thank you!

closed 2018-09-24 by rokusottervanger