cra-ros-pkg/robot_localization #418: What happens when an odom topic specifies another `child_frame_id` than `base_frame`

rokusottervanger, 2018-09-24:
I've been looking through the code (especially in the measurement callbacks and the `correct` method in the EKF implementation), and it seems that the `child_frame_id` in `odom` messages is not taken into account. 

What I think should happen is that the measurement model (converting state to measurement, as part of the `correct` step), should take into account the offset of the sensor using the transformation between the `base_frame` and the `child_frame_id` of the message.

Am I overlooking something, or is this indeed not implemented?
Thanks in advance!

rokusottervanger, 2018-09-24:
And what about the pose part of the odom message?

ayrton04, 2018-09-26:
It gets processed here:

https://github.com/cra-ros-pkg/robot_localization/blob/kinetic-devel/src/ros_filter.cpp#L1618

And we re-use its header, so it keeps the `frame_id`, which is what we want.

So, to summarize, in `r_l`, all pose data gets transformed from the sensor's reported frame to the `odom_frame` frame from your configuration. All velocity data gets transformed from the sensor's reported frame to the `base_link_frame` frame from your configuration. The exception to this rule is IMU data, but we're not talking about that here.

For the odometry message, we create a `PoseWithCovarianceStamped` message with the same `frame_id` as the odometry message. We copy the pose and covariance into it, and process it. We then create a `TwistWithCovarianceStamped` message, copy out the velocity and covariance data into it, and set its `frame_id` to the `child_frame_id` of the odometry message.

ayrton04, 2018-12-19:
Mind if I close this ticket?

rokusottervanger, 2018-12-20:
I still don't see how an off-center position measurement could be handled correctly with the code you pointed out to me. I see that poses are "prepared", meaning, among other things, they are transformed to the `base_link` frame. However, this cannot be done unless at least the robot's orientation is already known, which is not always the case. 

For example, an (extended) Kalman filter is neatly capable of inferring orientation from position and odometry data. If the position sensor is not aligned with `base_link`, rotation also implies translation of the position sensor. So in theory, turning on the spot would be enough to infer the orientation. I don't see how this would work using the code, as any position measurement is transformed into a "position measurement" of the `base_link` frame using the current estimate of the robot pose. Cuttnig the orientation out of the loop.

If I misunderstand, please correct me, but I think the measurement should not be transformed, but the state should be converted to the measurement representation (as usual, in a Kalman filter) using the measurement model. In this case, the measurement model should include the transformation between sensor and `base_link`, but the pose measurement should never be transformed.

ayrton04, 2018-12-20:
> I see that poses are "prepared", meaning, among other things, they are transformed to the base_link frame.

No pose data is transformed into the `base_link` frame. Only velocity data is. All pose data gets transformed from its `frame_id` to the EKF `world_frame` parameter (usually _map_ or _odom_). All twist data gets transformed from the `child_frame_id` (for nav_msgs/Odometry messages) or the `frame_id` (for TwistWithCovarianceStamped messages) to the `base_link_frame` EKF parameter (usually _base_link_).

Graphically, the filter assumes this:

```
  odom  <== Pose data (all pose data must have a direct transform to odom, and cannot be relative to base_link)
    |
    |  <-- (this is provided by the EKF itself)
    |
base_link  <== Twist data (all twist/velocity data must have a direct transform to base_link)
```

We just look up the transform from the source frame to the target frame, apply it to the source data, then fuse that into the filter.

So if we imagine an overhead camera system that measures the robot's XYZ position in 3D space, but is mounted facing, say, straight down, we may have to apply a transformation from the camera's world frame to the robot's world frame. That transform will be a simple rotation and translation. Once the data is in the correct frame, we can fuse it directly into the filter.

Reading back over your earlier questions and the most recent one, I think what you're trying to figure out is how to fuse pose data from a pose-providing onboard sensor that is not mounted at the robot's origin. That requires a bit more explanation.

Let's say we have some kind of pose measuring sensor onboard the robot, and it's mounted 0.5 meters left of the robot's centroid. Presumably, this offset is given by the _base_link_ -> _pose_sensor_ transform. This is an issue for `r_l`, because we assume all pose data has a direct transform available to the world frame (_odom_ or _map_), and is _not_ relative to _base_link_.  You're trying to measure world-frame data from a position that is relative to the robot's centroid, and therefore moving.

The problem is this: how can the EKF know whether the sensor is mounted on the robot or externally? If it's mounted externally and measuring the robot's pose in a world-fixed frame, then we just transform that data from its source from to, e.g., _odom_, and fuse it. If it's mounted on the robot, then we have to take the _base_link_->_sensor_ transform, rotate it by the robot's current orientation, then remove that offset from the robot, then fuse it.

Unfortunately, ROS pose message have only a `frame_id` field, so we can't know which transformation is appropriate. Transforming the state into measurement space won't fix the root issue, which is that we don't know what the transform should be.

FYI, we have to effectively [do this](https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/src/navsat_transform.cpp#L379) for `navsat_transform_node`, because the GPS may not be mounted at the sensor's origin.

> So in theory, turning on the spot would be enough to infer the orientation. 

Yes, I take your point on this, but given the generic nature of the filter, I think this is better handled within the sensor drive itself. Even if we changed the filter to automatically infer orientation from changes in position of an onboard pose sensor, we'd have to have some means of telling the EKF whether it _should_, because not everyone would desire that behavior.

In any case, we're working on moving towards a better state estimation package, so I'm doubtful that this one will undergo the fairly major changes required to support this use case. It may be wise to just make the sensor's transform relative to _odom_, and dynamically update it based on the robot's current orientation.

rokusottervanger, 2018-12-20:
Thanks for your extensive and patient answers! This clarifies things. I'll close the ticket.

closed 2018-12-20 by rokusottervanger