cra-ros-pkg/robot_localization #420: Gyro Only Orientation Localization Error

jwag, 2018-09-28:
Hello,

I think I may uncovered an issue with the way robot localization handles gyro data. I am using a Clearpath Jackal robot and the onboard imu which is mounted so that the x-axis faces towards the front of the robot, the y-axis to the left, and the z-axis up. Gyro and accell data is published on the topic /imu/data_raw. The orientation part of the message is set to all zeros with a -1 value for the first covariance.

We tilt the robot in the following way according the base_link:
Forward (+y)
Back to Level (-y)
Right (+x)
Back to Level (-x)
Turn Right 90 deg (-z)
Forward (+y)
Back to Level (-y)
Right (+x)
Back to Level (-x)
Turn Left 90 deg (+z)
Forward (+y)
Back to Level (-y)
Right (+x)
Back to Level (-x)

The resulting /odometry/filtered estimate works correctly before we rotate about the z. Then the tilting results an incorrect orientation. However, we we rotate back to the initial Z orientation the estimate behaves correctly again.

I have attached a bag file with /imu/data, /odometry/filtered, and /tf recorded. I have also attached my robot_localization.yaml file.
[gyro issue.zip](https://github.com/cra-ros-pkg/robot_localization/files/2429648/gyro.issue.zip)

ayrton04, 2018-10-02:
So I think this should most likely be a question on answers.ros.org, but at the very least, to investigate this, I'll need your full configuration file.

ayrton04, 2018-10-02:
By the way, for your test, you need to _also_ be fusing linear velocity data. If you only fuse angular velocity data into the filter, your covariance matrices are going to explode (see below), because you're not measuring X, Y, Z, or their velocities. When the covariance matrices explode, it can lead to strange behavior. Not saying that's what's causing this necessarily, but it needs to be corrected before I look at this. You can fake a `TwistWithCovarianceStamped` publisher with all zero values (make sure to set the `frame_id`s in that case).

```
header: 
  seq:       1984
  stamp: 
    secs: 1538063559
    nsecs:  834730387
  frame_id: "odom"
child_frame_id: "base_link"
pose: 
  pose: 
    position: 
      x: 0.00000000
      y: 0.00000000
      z: 0.00000000
    orientation: 
      x: 0.00153258
      y: 0.04461582
      z: 0.12024158
      w: 0.99174041
  covariance: [25197.5189, 126.556763, -1506.5567, 0.00000000, 0.00000000, 0.00000000, 126.556763, 25794.6695, 1465.96571, 0.00000000, 0.00000000, 0.00000000, -1506.5567, 1465.96571, 70256.0844, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.87387033, -0.0149394, 0.35134912, 0.00000000, 0.00000000, 0.00000000, -0.0149394, 0.89123394, -0.0390019, 0.00000000, 0.00000000, 0.00000000, 0.35134912, -0.0390019, 2.62323192]
twist: 
  twist: 
    linear: 
      x: 0.00000000
      y: 0.00000000
      z: 0.00000000
    angular: 
      x: -0.0124698
      y: -0.2889042
      z: 0.00103422
  covariance: [210.684183, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 210.684183, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 316.125730, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000012, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000012, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000000, 0.00000012]
```

jwag, 2018-10-02:
Ok. Thank you. I made a simple TwistWithCovarianceStamped publisher and set all values to 0 including the covariances. Should they be some small positive number instead? I set the frame_id to base_link which is coincident with imu_link.

I then modified the yaml file to include the linear velocity from the twist message. I then reran the same test which results in the same behavior as before. More specifically, after rotating -90 deg about Z when I tilt the robot forward (+Y) robot_localization says that it is tilting to the right ish (+X) and when I tilt the robot to the right (+X) robot_localization says that it is tilting to the back ish (-Y).

I have included the launch file, the config file, and the twist publisher script. What other files do you need?

Also, do you want me to move this over to a answers.ros.org?

Thanks for the help.
[gyro issue2.zip](https://github.com/cra-ros-pkg/robot_localization/files/2438639/gyro.issue2.zip)

ayrton04, 2018-10-11:
I just wanted to let you know that I haven't forgotten about this. I will get to it as soon as I can. Thanks!

ayrton04, 2018-11-27:
Confirmed that this is a bug. I know what's wrong. Will make a branch. Would you be willing to test it?

ayrton04, 2018-11-27:
PR submitted. Feel free to clone that branch and try it. The branch is relative to `melodic-devel`, but it ought to build in most environments. I'll backport the change to all relevant releases and bloom once it's merged.

nihad-s, 2018-11-28:
I tried the new branch with my IMU and now it finally works fine. The orientation calculated only from the angular velocities looks finally like the rotation vector internally calculated by the IMU. I have however sometimes discrete jumps in the orientation output but it might be something else as I don't have any other input as of right now except the angular velocities.

ayrton04, 2018-11-28:
Oh? As this is new code, I'd be interested in seeing the discontinuities, if it's not too much trouble.

nihad-s, 2018-11-28:
[loca_discrete_jumps.bag.zip](https://github.com/cra-ros-pkg/robot_localization/files/2625060/loca_discrete_jumps.bag.zip)
I am not sure if this is a good format to show my results. I recorded in my bag the input IMU topic of the ekf node and the generated tf and filtered/odometry. I used a very simple launch file for this. Really severe discontinuities are at around 50s and 1.15s where the odom jumps wildly around. But sometimes there also smaller jumps which are harder to notice. 

<node pkg="robot_localization" type="ekf_localization_node" name="ekf_se_odom" clear_params="true" output="screen">
    <param name="imu0" value="/base/imu"/>
    <param name="imu0_differential" value="false"/>
    <param name="imu0_relative" value="false"/>
    <param name="imu0_remove_gravitational_acceleration" value="false"/>
    <param name="print_diagnostics" value="true"/>
    <param name="imu0_queue_size" value="10"/>
    <rosparam param="imu0_config">[false, false, false,
                               false, false, false,
                               false, false, false,
                               true,  true,  true,
                               false, false, false]</rosparam>
    <remap from="odometry/filtered" to="base/filtered_odom"/>
  </node>

nihad-s, 2018-11-28:
Yes, that's the only data I am fusing in that example, reducing the process noise covariance values would probably make it work a little longer working, but eventually something like that is supposed to happen without correction. I was just not sure if that was the problem or something else so wanted to make sure and ask.

closed 2018-12-19 by ayrton04