cra-ros-pkg/robot_localization #427: datum service not running in melodic

arnaudpenfornis, 2018-11-22:
We have recently upgraded our project from kinetic to melodic and we are facing an issue.
The datum service worked fine on kinetic and now it is not even running.
We have checked the diff between the two branches and we can't figure out what change can cause this. 

We tried on our melodic project:
- Build the kinetic branch, it worked.
- Build the melodic branch, no error are shown and the nodes are launched but the datum service is not running.

Have you an idea of what changed between the two branches ?

Regards

ayrton04, 2018-11-22:
Not sure. The service is still there:

https://github.com/cra-ros-pkg/robot_localization/blob/melodic-devel/src/navsat_transform.cpp#L95

Have you verified this on different machines? Any chance you can try running from binaries?

schadocalex, 2018-11-22:
I find why, here's the PR:
https://github.com/cra-ros-pkg/robot_localization/pull/428

ayrton04, 2018-11-23:
Ah, this is a side effect of the move to nodelets.

We used to declare the datum service server locally, because it was in the main `run` method:

https://github.com/cra-ros-pkg/robot_localization/blob/81f7d9b4f88165567ef433c7ca75cb6042fa46c1/src/navsat_transform.cpp#L105

But for Melodic, there was a PR to move to nodelets, and that service moved to the constructor, which obviously doesn't work. Nice catch, @schadocalex.

closed 2018-11-27 by ayrton04