cra-ros-pkg/robot_localization #434: Failed to build robot localization code by sourcing  head/master ros2 branch

poonam1120, 2018-12-14:
Required Info:
    Operating System: Ubuntu 18.04 
    ROS2 :- master/head branch

Bug report:- 
I had imported ros2 branch of robot localization (https://github.com/cra-ros-pkg/robot_localization/tree/ros2)
and tried to build by sourcing ros2 latest release branch (as per migration guide),it worked fine.
But when I try to build the same by sourcing master/head branch of ros2, colcon build failed throwing below errors.

Error Info:- 

> /home/user/master-try-RL/src/robot_localization/src/ros_filter.cpp: In constructor ‘robot_localization::RosFilter::RosFilter(rclcpp::Node::SharedPtr, robot_localization::FilterBase::UniquePtr&)’:
> /home/user/master-try-RL/src/robot_localization/src/ros_filter.cpp:74:30: error: no matching function for call to ‘tf2_ros::Buffer::Buffer()’
>      filter_(std::move(filter))
>                               ^
> In file included from /home/user/master-try-RL/src/robot_localization/include/robot_localization/ros_filter_utilities.hpp:39:0,
>                  from /home/user/master-try-RL/src/robot_localization/src/ros_filter.cpp:35:
> /home/user/master_ros2_ws/install/tf2_ros/include/tf2_ros/buffer.h:65:20: note: candidate: tf2_ros::Buffer::Buffer(rclcpp::Clock::SharedPtr, tf2::Duration, bool)
>      TF2_ROS_PUBLIC Buffer(rclcpp::Clock::SharedPtr clock, tf2::Duration cache_time = tf2::Duration(tf2::BUFFER_CORE_DEFAULT_CACHE_TIME), bool debug = false);

***********
I suspect that the code should be changed as per the master/head branch of ros2.

Thanks

ayrton04, 2018-12-19:
Have you tried it against the ROS Crystal binaries?

poonam1120, 2018-12-20:
Hi Tom,
After adding one line of code as above , some of my test cases are failing due to timing issue. 
Do you have any idea/suggestion if that one line is affecting the timing setup.
Thanks.

ayrton04, 2019-01-09:
Once [this](https://github.com/cra-ros-pkg/robot_localization/pull/436) is merged, can you try again?

SteveMacenski, 2019-12-04:
I assume this has been resolved by the dashing branch?

closed 2021-05-20 by SteveMacenski