cra-ros-pkg/robot_localization #441: Document undocumented parameters

maracuya-robotics, 2019-01-17:
Some parameters like `predict_to_current_time` (and maybe more?) are not documented.

ayrton04, 2019-01-17:
Thanks for the report. I'd welcome a PR!

maracuya-robotics, 2019-01-17:
Wow, that was fast!

Found another one, `disabled_at_startup`. IMHO documentation should be done by someone who knows what these parameters do. I can just guess by the name and by looking at the code.

diasdm, 2019-02-14:
I think these are all the undocumented parameters for the documented nodes.

In `navsat_transform.rst` is missing:
* transform_timeout (only documented for `state_estimation_nodes.rst`)

In `state_estimation_nodes.rst` is missing:
* base_link_frame_output
* predict_to_current_time
* disabled_at_startup

seebq, 2019-03-31:
Hey @maracuya-robotics and @diasdm I submitted a PR #454 for y'all.  Can you review and let @ayrton04 know if this documents the parameters that were missing?

ayrton04, 2019-04-26:
Assuming this is addressed by #454. Thanks @seebq.

closed 2019-04-26 by ayrton04