cra-ros-pkg/robot_localization #442: Inaccurate initialisation of utm->map causing inaccurate gps goal pose estimations

GrantDare, 2019-01-18:
I am using this package to localise a robot for outdoor navigation. My issue is that when I use the _RobotLocalization::NavsatConversions::LLtoUTM_ to convert a GPS point to UTM and then listen to the utm->map transform provided by the navsat transform node to get a map frame goal, the goal is always wildly inaccurate by a random amount (both distance and heading from robot). The odometry pose of the map frame and the odom frame is perfectly aligned which tells me that the _magnetic declination_ and _yaw offset_ params are correct. 

This leads me to believe there is an issue in the initialisation of the node. Is there a recommendation to make this more accurate or perhaps plans to implement a filter on initialisation due to the discrete nature of GPS data?

Thank you in advance!

ayrton04, 2019-01-18:
So before we suggest that this is a bug, I think it should be a question on answers.ros.org. When you ask it, please include the code you are using to transform the data, and then direct me to the question here.

Offhand, it sounds like you may be using the inverse of the transform you actually want.

Once we get the `utm->map` transform, every single GPS point that we receive is then (a) converted to UTM coordinates, and (b) passed through the transform, resulting in an updated `/odometry/gps` position. If there was an issue with the `utm->map` transform, then it would follow that every GPS reading after the first one would _also_ be wildly inaccurate, and the output of `navsat_transform_node` would jump all over the place.

In rviz, the odom and map frame positions will _always_ perfectly align (assuming you are using `r_l` EKF or UKF instances for both), because the second EKF is constantly updating its map->odom transform. They'll always be consistent.

GrantDare, 2019-01-18:
Thank you @ayrton04  for the speedy response. I have had a [question](https://answers.ros.org/question/312848/navsat_transform_node-utm-map-transform-is-highly-inaccurate/) up on ROS answers and have just modified it to include some further information to assist you :)

closed 2019-04-26 by ayrton04