cra-ros-pkg/robot_localization #445: broadcast_utm_transform_as_parent_frame not found in documentation

diasdm, 2019-02-11:
Hello,
As far as I know, the parameter broadcast_utm_transform_as_parent_frame is used in [r_l](https://github.com/cra-ros-pkg/robot_localization/blob/f6bf556407945e7b192c78959c51d2f10f62e488/src/navsat_transform.cpp#L84), but it is not mentioned in the [documentation](http://docs.ros.org/melodic/api/robot_localization/html/state_estimation_nodes.html). Shouldn't it be?
It is already described [here](https://github.com/cra-ros-pkg/robot_localization/blob/f6bf556407945e7b192c78959c51d2f10f62e488/params/navsat_transform_template.yaml#L28), so a similar description can be used.
Thank you.

ayrton04, 2019-02-11:
Agreed. Care to submit a PR?

diasdm, 2019-02-11:
Done, #446

closed 2019-02-13 by diasdm