cra-ros-pkg/robot_localization #450: tf2  issue with lidar odometry integration

BrunoEduardoCSantos, 2019-02-27:
Hi,
Running both robot_localization package and the slam ros package that generates lidar odometry, the following error comes up:

    ```
    at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument "/camera" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:
         at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument "/camera_init" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:
         at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument "/camera" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:
         at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument "/camera_init" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:
         at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
Warning: Invalid argument "/camera" passed to canTransform argument source_frame in tf2 frame_ids cannot start with a '/' like:
         at line 134 in /tmp/binarydeb/ros-kinetic-tf2-0.5.20/src/buffer_core.cpp
```
My tf tree is the following:

![rosgraph](https://user-images.githubusercontent.com/3111976/53488750-21963980-3a87-11e9-94a5-266882665725.png)

Can anyone help me?
Thanks,
Bruno

BrunoEduardoCSantos, 2019-02-27:
I just need to remove the slash in SLAM ros package.

closed 2019-02-27 by BrunoEduardoCSantos

ayrton04, 2019-02-27:
Yeah, `tf2` doesn't support leading slashes. If you have further questions/issues, answers.ros.org is a great resource.