cra-ros-pkg/robot_localization #451: Confusion about the update logic in the function "FilterBase::processMeasurement()"

zhangjian-neu, 2019-03-01:
In the function "FilterBase::processMeasurement", lines 223-231, the codes say "Only want to carry out a prediction if it's forward in time. Otherwise, just correct."

Why the filter should correct when delta <= 0, i.e., measurement.time_ <= lastMeasurementTime_???

That is really confused. From my view, in such a case, the filter should not predict or correct. The filter  should just discard the measurement.

ayrton04, 2019-03-19:
Imagine you have a high-frequency measurement source like an IMU, and a low-frequency measurement source that has some lag in its processing (e.g., visual odometry). You process an IMU measurement in one cycle, and its time is `t = 1`. At the same time, your visual odometry node starts processing the latest image frame. At time `t = 1.05`, you get a new IMU measurement. At time `t = 1.1`, your visual odometry node publishes a measurement with time stamp `t = 1`. That measurement is old, so if we follow your suggestion, we'll just drop that measurement. In fact, we'll pretty much never use the visual odometry.

This logic has been around since the early days of `r_l`, but then we had users contribute a handy history feature, whereby we could rewind the state of the filter to before a given measurement, insert that measurement, and "replay" all measurements up to the current time. I'd recommend using that, but in the meantime, the logic as it stands has value.

closed 2019-03-19 by ayrton04

zhangjian-neu, 2019-03-27:
I get it. Really thanks for your kindly response.