cra-ros-pkg/robot_localization #456: imu  frame prepare

GITSHOHOKU, 2019-03-28:
Hi  
I 'm dealing with the  imu frame_id in urdf . I follow the tutorial here [prepare data](http://docs.ros.org/melodic/api/robot_localization/html/preparing_sensor_data.html)
It said can make a static transform between  base_link and imu_frame, but Imu still broadcast
it's tf  in a fixed(mine is odom) frame,Did it mean I should keep both of them in my urdf file or I should remove the fix tf ,and how to remove it ?

Thanks and I can't ask it on ros.org becaus of it use link with google which I can't connect?

GITSHOHOKU, 2019-03-28:
To simplify the question
Should I keep both two transform：
1. static TF between  base_link  and  imu-mount-position  
2. imu own TF  wrt  odom(or map) which is fixed to provide  orientation
then in my urdf there always two different  tf  of  imu_data's  same  frame_id

ayrton04, 2019-03-28:
Thanks for the question. However, we ask that you please direct all questions to answers.ros.org.

closed 2019-03-28 by ayrton04