cra-ros-pkg/robot_localization #457: Support more/custom/pluggable motion models

beetleskin, 2019-04-10:
Currently, the filter work on the 6DOF omni-directional motion model exclusively. 
In order to improve the localization of robots where this motion model is not applicable, the user should be able to select from different motion models, or be able to provide a custom motion model, e.g. via the plugin-lib.

Rationale:
Most robots do not have 6DOF, only drones and similar do actually. Often the robot is wheeled or legged, meaning it can't fly. There are omni-directional robots, differential-drive robots, ackermann-based robots, swimming robots .. It is assumed that the localization accuracy will benefit from more applicable motion models.

ayrton04, 2019-04-26:
Yeah, this is a common request. The reason it hasn't been addressed is primarily because we're hoping to release an `r_l`-compatible package that is built on top of `fuse`:

https://github.com/locusrobotics/fuse

`fuse` will have support for easily adding new kinematic models to the package.

In the meantime, however, you can actually get the behavior of other kinematic models within `r_l`. For example, if you use `two_d_mode` and also feed 0 measurements for `Y` velocity from your sensor data, you end up, IIRC, with the same math you get for the unicycle model. This package gets used on thousands of 2D differential drive robots 24/7 in commercial environments, as well as on some 2D omnidirectional robots. Granted, this is less computationally efficient that a dedicated 2D model, but the package is pretty lightweight regardless.

ayrton04, 2019-04-26:
Oh, and if you increase your process noise, you can at least have a means of representing the kinematic model error if you have, e.g., an Ackermann-steered robot.

beetleskin, 2019-04-29:
Wow, rebuilding this on top of fuse sounds like a bigger operation. Is there a document on this proposal one can dive into?