cra-ros-pkg/robot_localization #458: robot_localization node stops publishing /odometry/filtered after a while

RajathSwaroop, 2019-04-24:
I have been using robot_localization package to fuse data from wheel odometry, IMU and another source of odometry. The robot_localization node publishes data for a while and stops publishing on /odometry/filtered after like 2 or 3 hrs even though data on the other 2,3 topics are being published at the required frequency. Unsure what could be the cause for this behavior.

ayrton04, 2019-04-25:
This is most likely an issue with your sensor data (or your platform). This package runs on thousands of robots 24/7 in commercial environments, so I'm guessing you might have some bad covariance data creeping in.

However, I can't really comment unless you give far more detail, like

- What version of the package are you using? Are you building from source or using binaries?
- On what platform are you running it?
- Did you create the sensor data sources, or are they from another open source package?

Please provide sample inputs from all input sources. Even more helpful would be a bag of data that contains a period in which the EKF stops publishing.

AlexisTM, 2019-05-03:
On top of that, I would recommend checking the diagnostic topic.

ayrton04, 2019-07-01:
Closing. Feel free to ask this on answers.ros.org.

closed 2019-07-01 by ayrton04