cra-ros-pkg/robot_localization #469: How does imu fuse?

huchhong, 2019-06-19:
I have read code of robot_localization and am confused of how imu is fused. It seems to me that if differential is set to false, then how can imu data in some world fixed frame to be fused with like odometry?

I also post a question at ROS Answer: http://answers.ros.org/question/326236/robot_localization-how-to-deal-with-imu-data/, please help me understand this.

ayrton04, 2019-06-19:
Thanks for the question. ROS Answers is indeed the preferred method for asking questions, and I will get to it soon.

closed 2019-06-19 by ayrton04

huchhong, 2019-06-20:
Sorry to bother again. Can you help to answer my question on ROS Answer?

ayrton04, 2019-06-20:
I'm afraid my work is very busy at the moment. Please be patient, and I will answer it as soon as I can. Answering questions is often time consuming, and I don't have a lot of spare cycles at the moment to dedicate to it.