cra-ros-pkg/robot_localization #471: navsat frame ID leading slash

lstromeyer, 2019-06-28:
I am unsure if I should be filing this ticket here. The issue I encountered is that the NMEA serial package I used the standard ROS libnmea_navsat_driver (specifically the function driver.get_frame_id()). That I believe based on the old tf transform library automatically put a leading slash to frame ID which tf2 does not like. Either someone did not update the libnmea_navsat_driver or they kept it that way for backwards compatibility. Your navsat_transform_node did not like this. Would it be possible to add a check in your package to strip any leading slashes? Your package is not the root of the source but I believe a similar issue might come from a number of sources. I believe you already do this for localization nodes, so can this be added to the navsat transform node?
I hope this is useful and I hope I did not get anything wrong here when tracing the problem.

ayrton04, 2019-07-01:
Sounds reasonable. Have you opened a ticket on the `nmea_navsat_driver` repo as well?

Also, this is a pretty trivial change, and if you're comfortable doing so, I'd be willing to accept a PR for it.

lstromeyer, 2019-07-02:
Thanks for the reply. I'll make the branch with the changes sometime in the
next few days.
I haven't yet done it on the other repo but it is on my to do list.

closed 2026-11-02 by ayrton04