cra-ros-pkg/robot_localization #473: Port to ROS2 Dashing

rotu, 2019-07-10:
There should probably be a ROS2 Dashing branch in this repo.
Feel free to pull from our fork https://github.com/RoverRobotics-forks/robot_localization

ayrton04, 2019-07-26:
Is your branch based on the Bouncy/Crystal branches on this repo? I really need to clean up the ROS2 branches here. I want to remove support for diagnostics until it's been officially released for ROS2.

rotu, 2019-07-26:
My branch is based on master. The diagnostic publisher seems to be working pretty solid.

ayrton04, 2019-07-26:
Sorry, so you branched the ROS1 version (we have no `master` branch) and also ported it to ROS2? I'm only asking because I made a ROS2 branch for Bouncy, and then we had a user contribution to get the tests running and re-establish the diagnostics.

rotu, 2019-07-26:
Oops. Yes, we branched from ROS1. Please open a dashing-devel branch. I'll rebase my changes on it and submit a PR.

ayrton04, 2019-08-08:
A few things.

First, thanks for taking the time to port to ROS2 Dashing. Is there anything about Dashing that would make the code significantly different than Crystal? I only ask because I (and others) have already ported this package to ROS2, and we have `bouncy-devel` and `crystal-devel` branches. When I originally did the ROS2 port, I also made changes to some code style and a few other things, and I'd like to retain them. I'd rather not have our Bouncy and Crystal branches come from our changes, and the Dashing branch come from a completely different source, if that makes sense.

I do need to make some major changes to the Crystal release, though. It's using diagnostics, which, as of Crystal, was not released as a binary package. So I'd welcome PRs to get things cleaned up.

Finally, I should note that I am not actively using ROS2, so I have limited cycles to devote to working on it. I'd really like to get the tests moved over to a proper ROS2 framework, rather than the complex setup they have now.

rotu, 2019-08-08:
Yes, the biggest change for dashing is that you need to declare all your parameters or things will crash. I understand the want to keep the version branches as similar as possible. I really just need a base branch to submit a PR on.

rotu, 2019-08-08:
@ayrton04 I rebased my changes in https://github.com/RoverRobotics-forks/robot_localization

ayrton04, 2019-08-09:
https://github.com/cra-ros-pkg/robot_localization/tree/dashing-devel

mabelzhang, 2019-09-10:
There seems to be a lot of diffs ([156 commits](https://github.com/cra-ros-pkg/robot_localization/compare/dashing-devel...melodic-devel)) between the `melodic-devel` and `dashing-devel` branches. Is this expected? If not, what are the functionalities / features in Melodic that aren't in Dashing?

rotu, 2019-09-17:
@mabelzhang the changelist is a good place to start. https://github.com/cra-ros-pkg/robot_localization/compare/dashing-devel...melodic-devel#diff-de0a9b67ffe2874a076b5fdb15bad484

@klintan, you were also working on porting this forward to Dashing. Are you still doing that or should I?

klintan, 2019-09-17:
> @klintan, you were also working on porting this forward to Dashing. Are you still doing that or should I?

It stalled because of geography_info not being released :( But @SteveMacenski is working on getting that one done almost as we speak. 

https://github.com/ros-geographic-info/geographic_info/pull/28

So as soon as that is done the work can proceed. But I saw that https://github.com/cra-ros-pkg/robot_localization/pull/487 contained som very similar changes, so we'll have to try to sync that. I don't object to you taking over :D

mabelzhang, 2019-09-24:
> @mabelzhang the changelist is a good place to start. [dashing-devel...melodic-develdiff-de0a9b67ffe2874a076b5fdb15bad484](https://github.com/cra-ros-pkg/robot_localization/compare/dashing-devel...melodic-develdiff-de0a9b67ffe2874a076b5fdb15bad484)

Yeah... that was the same link I posted :P While many of the diffs are builds specific to Melodic, there are new features like [lat/long conversion service](https://github.com/cra-ros-pkg/robot_localization/commit/fbd3a72d344fc9e63085ca489bb16f31b2daf3a1), [meridian convergence adjustment](https://github.com/cra-ros-pkg/robot_localization/commit/2b57df66ded98b3ae4d08176f54beead4e9e1f5f), [variable renaming](https://github.com/cra-ros-pkg/robot_localization/commit/17d4f100a46e7fbed9c17c874f6f9baa3091f54f), [const& change](https://github.com/cra-ros-pkg/robot_localization/commit/c1963c34868af0d3d0699c0928fa2185fee72106), etc. that might want to make it into Dashing.

We are thinking the best way might be to squash and rebase the current `dashing-devel` on `melodic-devel` so that those new features get into Dashing. Any objections to that?

@ayrton04?

ayrton04, 2019-09-25:
No objections as such, but I changed the coding standards for the ROS2 branch(es), e.g., variable names are `snake_case` now. I think that rebasing is going to create a world of pain for whoever takes it on, and I wonder if cherry picking commits and then cleaning them up, one at a time, would be easier.

We should probably also weigh the value of the features in melodic against the effort in porting them. It's not clear to me how many users require some of the features that were added, so the return on the investment of time might be small.

mabelzhang, 2019-12-04:
As a heads up, we cherry picked some (a lot of) commits in the diff above https://github.com/cra-ros-pkg/robot_localization/compare/dashing-devel...melodic-devel to port to dashing-devel.

The major one seems to be the 71 commits from 16 Oct 2017, which adds robot_localization_estimator, ros_robot_localization_listener, ros_robot_localization_listener, and tests. I'm in the process of doing essentially
```
$ git cherry-pick 11e0da9^..e912e40
```

If there's any reason that these classes shouldn't be ported to Dashing, e.g. nobody uses them, they're irrelevant for some reason, etc., please let me know. Otherwise I'll continue to port them over and open a PR once it's done.

@SteveMacenski just tagging because maybe you know stuff.

SteveMacenski, 2019-12-04:
can't say I've used that bit. @ayrton04 is probably the best resource on whether that's generally useful or useful to what domain of people

SteveMacenski, 2019-12-09:
@mabelzhang I'm going to leave the porting of those commits -- I was going to add back some of the services but that would be in the scope of what you're working on and I don't feel like making you deal with merge conflict hell. One of the services that I feel strongly about is the toggle filter service.

mabelzhang, 2019-12-09:
Thank you! I'm a little behind on this. I started the cherry pick but it got put aside. I hope to get back to it next week.

I did notice the toggle filter service. I think that is farther down the line. The range above is taking me a while, so if you are merging in the toggle filter, it might decrease the diff list and take it off my list of things.

SteveMacenski, 2019-12-10:
@mabelzhang I don't suppose you've tried to compile this in Eloquent? I get a bunch of linking issues that is leaving me scratching my head. As far as I can tell from all my other Eloquent code in navigation this is linking exactly the same. 

Got it -- well lets wait for the existing PR to get looked over and merged before I start messing with anything else.

JWhitleyAStuff, 2020-01-19:
@SteveMacenski / @mabelzhang / @ayrton04 - What is the status of the port? How can I help? I need to make sure this is released ASAP for a project that I'm a part of but I need to know what's been done and what hasn't to know where I can be of use.

Also, @ayrton04 - I noticed there is no CI on this repo, on any of the branches. I would be happy to set it up with whatever service you like (though I usually use CircleCI) if you would allow me the permission.

SteveMacenski, 2020-01-19:
I cleaned up some of the repo and ported in things from the ROS2 forks relevant. 

Mabel has PR s open cherrypicking commits from ROS1 over to ROS2. I have approved them but I want a second look from someone deeply familiar with the project. 

I haven’t been able to get the dashing branch to build in Eloquent. Its been a couple months since I’ve had cycles to revisit it. If nothing, Eloquent stuff if that’s important to you would be appreciated. 

10-4 on CI. I think we discussed that in another thread. Its not in the permissions I have to do it for you. We’ll need Tom.

JWhitleyAStuff, 2020-01-19:
I'm mostly interested in getting Dashing released for now. I'll keep an eye on the PR. After those changes are merged, would you say it is ready for a 1.0 Dashing release? I'm happy to add linting if necessary. ;)

SteveMacenski, 2020-01-19:
I would say so, yes.

Linting always appreciated ;-)

ayrton04, 2020-01-19:
> Also, @ayrton04 - I noticed there is no CI on this repo, on any of the branches

I mean, we have the [ROS1 build farm jobs](http://build.ros.org/search/?q=robot_localization&Jenkins-Crumb=d4e78d86e86fb910042f30c5ad9e4600a7b0fa11c99a92c2ef14634825a622f1) running the package's `gtest`s and `rostest`s for every PR and commit (e.g., http://build.ros.org/job/Mpr__robot_localization__ubuntu_bionic_amd64/18/). It's integrated with GH so that PRs can be checked before they are merged. That stuff runs on all the ROS1 branches. I don't think we've gone through the bloom process for ROS2 yet, though.

SteveMacenski, 2020-01-21:
PRs are in. @mabelzhang should we expect any more cherry picks. If so, how many?

mabelzhang, 2020-01-21:
Yes. I currently have 2.5.2 to 2.6.1, with 2.6.1 close but not done yet. ~15 commits.

I'm really just porting all the diffs between melodic and dashing unless I see stuff have already been ported, which is few. I'm counting ~25 commits of real content.

closed 2020-02-05 by SteveMacenski