cra-ros-pkg/robot_localization #475: Navsat Transform initial computation of initial transform done with ros::Time(0) and not msg->stamp

samkys, 2019-08-03:
I have been fighting Navsat transform node a bit and before I asked too many questions I wanted to dig through the source.

Is there a reason that this is done with the current time and not the GPS stamped time? As I would assume there is a possibility of a offset especially if whoever is providing the `/fix` message to Navsat transform is delayed with respect to the actual GPS data received.

First called here: 
```
//  void NavSatTransform::run()
  void NavSatTransform::periodicUpdate(const ros::TimerEvent& event)
  {
    if (!transform_good_)
    {
      computeTransform();
```

Which then calls:
```
      if (!use_manual_datum_)
      {
        getRobotOriginUtmPose(transform_utm_pose_, transform_utm_pose_corrected, ros::Time(0));
      } 
```

Which then calls:
```
  void NavSatTransform::getRobotOriginUtmPose(const tf2::Transform &gps_utm_pose,
                                              tf2::Transform &robot_utm_pose,
                                              const ros::Time &transform_time)
  {
    robot_utm_pose.setIdentity();

    // Get linear offset from origin for the GPS
    tf2::Transform offset;
    bool can_transform = RosFilterUtilities::lookupTransformSafe(tf_buffer_,
                                                                 base_link_frame_id_,
                                                                 gps_frame_id_,
                                                                 transform_time,
                                                                 ros::Duration(transform_timeout_),
                                                                 offset);
```

And as you can see, it uses the transform time passed in which is ros::Time(0) and not the GPS message stamp. Is there a reason for this method of doing it or what is the reasoning because to me it seems as we can introduce a slight shift in the transform.

ayrton04, 2019-08-03:
A couple things:

1. `computeTransform` is only executed once in the entire lifecycle of the node. Once the UTM transform has been computed, none of that code will be run again. So, even if the transform in question were sensitive to timing issues, it would just be for one cycle. But...
2. ...the time stamp is used, as you posted, to look up a transform from the robot's `base_link` frame to the robot's `gps` frame. In other words, it's just looking up the mounting position of the GPS sensor on the robot. Unless your GPS is somehow moving around on your robot (and I've yet to see it!), that transform will be static, e.g., a sensor that is mounted 0.1 meters in front of the robot's centroid will _always_ be mounted 0.1 meters in front of the robot's centroid.

That method is just taking the UTM pose that we got from the raw GPS data, and adjusting it so we get the UTM pose of the robot's centroid (its origin, hence the method name).

If you have further questions, I'd ask you to please direct them to answers.ros.org. Thanks!

samkys, 2019-08-04:
Thanks for the quick response!

Sorry about posting this here and now I feel a little dumb for posting the question/issue after reading your response and the source again. Sure enough, just didn't catch that previously.

closed 2019-08-04 by samkys

ayrton04, 2019-08-05:
No worries whatsoever. Feel free to ask more questions on answers.ros.org. I have been on vacation for a bit, but will try to get over there to answer some questions soon.