cra-ros-pkg/robot_localization #477: ROS2 Beginner question: How to start navsat_transform_node ?

klintan, 2019-08-05:
Hey! 
So I have what is most likely an beginners question to say the least. However I can't find any way to start the navsat_transform_node. I've read through the tutorial in ROS1 and googled a bunch and it seems most people just put down the package name `robot_localization` and node executable `navsat_transform_node`. 

Now I've checked my `ros2 pkg executables` and the only available node in the robot_localization package is `se_node` (state estimation node I presume). Checking the `CMakeLists.txt` file, there is also no executable or rclcpp component being installed in the `lib ` folder so I really don't understand/think that that is the way I'm suppose to start it. 

Any advice is appreciated! My next step would be to actually add the `navsat_transform_node` in the cmakelists to make it an executable so that it can be run from the ros2 run cmd. But I think I'm just missing something fundamental (Because I can't find the executable in the ROS1 branches either)

Many thanks!

klintan, 2019-08-06:
Looking at the CMakeLists.txt file in the ROS1 branch, it's actually in the cmake file, meaning I'm not going crazy, it's just not implemented/activated yet in the ROS2 port. I'll put in a PR.

closed 2019-08-06 by klintan

ayrton04, 2019-08-07:
Just a quick note: `geographic_msgs` was not released at the time we ported this to ROS2, hence the disabled `navsat_transform_node`.