cra-ros-pkg/robot_localization #481: Configuring your Package

oconnor127, 2019-08-18:
Hey,

i've got a serious problem by setting up your package. At first, i am using Windows 10 and Ros-Melodic but maybe that problem is independent of my OS (i think so because thats my first usage of external package from github). 

I've downloaded the Package from here and put in in my workspace in here: D:\catkin_ws\src (parallel to my own other packages). If i'm going to build that now i've got some problems that "geographic_msgs" and "uuid_msgs" is unknown. So what i've done is to copy that msgs-packages from the ros website into the src folder of catkin_ws. (in catkin_ws/src there are 3 folders (geographic_info-master,robot_localization-melodic-devel,unique_identifier-master) and a CMakeLists now). 

The Error disappears. Now the building process runs till 64% and the following error occurs:

[ 64%] Building CXX object robot_localization-melodic-devel/CMakeFiles/filter_utilities.dir/src/filter_utilities.cpp.obj

cl : Command line error D8021 : invalid numeric argument '/Werror' NMAKE : fatal error U1077: "C:\PROGRA~2\MIB055~1\2019\COMMUN~1\VC\Tools\MSVC\1422~1.279\bin\Hostx64\x64\cl.exe": Return-Code "0x2" Stop.

NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\bin\HostX64\x64\nmake.exe"": Return-Code "0x2" Stop.

NMAKE : fatal error U1077: ""C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\bin\HostX64\x64\nmake.exe"": Return-Code "0x2" Stop. Invoking "nmake" failed

Do you know how to solve that error? Was that the right way to install your package(I cant use sudo apt-get and the win-package manager chocolatey doesnt offer that package so i tried this way)? I would appreciate your help.

Thanks in advance

ayrton04, 2019-08-19:
So I'm going to guess that the issue is indeed related to the fact that you're using Windows; if the package wasn't building for people in Linux, I can promise you I'd have heard about it. In any case, [the release is building fine on the build farm](http://build.ros.org/job/Kbin_uX64__robot_localization__ubuntu_xenial_amd64__binary/6/).

The first build issue was just that you didn't install the package dependencies (in this case, `geographic_msgs` and `uuid_msgs`) before you tried to build it.

I can't comment on the second error, because it doesn't appear to point to any line number. I don't use Windows, I'm afraid, so I think your best course of action is to ask on [answers.ros.org](answers.ros.org).

oconnor127, 2019-08-19:
ah ok yepp i just assumed i made a mistake (in some configuring stuff like choose the correct path were to unzip your code) because i'm quite new to ros. I already asked there but unfortunately nobody answered. Thank you anyway.

closed 2019-08-27 by ayrton04

oconnor127, 2019-08-27:
Forget to close that issue. It works one windows. For everyone who troubles with the same: https://answers.ros.org/question/330894/install-robot_localization-package-issues/