cra-ros-pkg/robot_localization #484: Incorrect IMU make EKF is not good ??

Fanny-one, 2019-08-26:
Now i'm using robot_localization to fuse imu, gps and odom. 
mems integrated navigation system provides imu and gps, like this picture.
[image]
It is mounted by right-front-up.
mems integrated navigation system can provide roll, yaw ,pitch ,so I read them to IMU' orientation instead of magnetometer and modify magnetic_declination_radians 0 from navsat_params.yawl. Yaw offset is also 0.
My robot_localizaton result is as below.
[image]
The blue , yellow and red lines represent  "odom" ,"odometry/gps" and "odometry/filtered". But I don't know why?

Fanny-one, 2019-08-26:
<node pkg="tf2_ros" type="static_transform_publisher" name="bl_imu" args="0 0.0 0.85 -1.570796327 0 0 base_link imu_link" />

  
  <node pkg="tf2_ros" type="static_transform_publisher" name="gps_transform" args="0 0 0.8 0 0 0 base_link gps_link" />

Fanny-one, 2019-08-26:
`
  <node pkg="tf2_ros" type="static_transform_publisher" name="bl_imu" args="0 0.0 0.85 -1.570796327 0 0 base_link imu_link" />

  
  <node pkg="tf2_ros" type="static_transform_publisher" name="gps_transform" args="0 0 0.8 0 0 0 base_link gps_link" />`

ayrton04, 2019-08-26:
Thanks for the question. However, this should really be directed to answers.ros.org. When you ask the question, please include your EKF config, as well as sample messages from every sensor input.

closed 2019-08-26 by ayrton04

Fanny-one, 2019-08-26:
Thank you! I will repeat the question .