cra-ros-pkg/robot_localization #485: Fuse GPS with IMU question

oconnor127, 2019-08-27:
Hey Tom,

i've got a stupid question. I'm trying to fuse the IMU data with GPS data. I've tried to stick to your tutorial but got the following question: Is that the right way to subscribe to the topics? The Data which are published by the topic "odometry/filtered/NAVI" looks quite strange.. In the following I'm just showing the most important details (topic subscribing) of my launch file..

<launch>
<node name="test_ekf_localization_node_bag1_ekf" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="imu0" value="/kitti/oxts/imu"/>
</node>

<node name="ekf_map" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="odom0" value="/odometry/gps"/>
<param name="imu0" value="/kitti/oxts/imu"/>
<remap from="/odometry/filtered" to="/odometry/filtered_NAVI" />
</node>

<node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true">
    <remap from="/imu/data" to="/kitti/oxts/imu" />
    <remap from="/gps/fix" to="/kitti/oxts/gps/fix" />
  </node>

</launch>

oconnor127, 2019-08-27:
<launch>
<node name="test_ekf_localization_node_bag1_ekf" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="imu0" value="/kitti/oxts/imu"/>
</node>

<node name="ekf_map" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="odom0" value="/odometry/gps"/>
<param name="imu0" value="/kitti/oxts/imu"/>
<remap from="/odometry/filtered" to="/odometry/filtered_NAVI" />
</node>

<node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true">
    <remap from="/imu/data" to="/kitti/oxts/imu" />
    <remap from="/gps/fix" to="/kitti/oxts/gps/fix" />
  </node>

</launch>`

oconnor127, 2019-08-27:
<launch>
<node name="test_ekf_localization_node_bag1_ekf" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="imu0" value="/kitti/oxts/imu"/>
</node>

<node name="ekf_map" pkg="robot_localization" type="ekf_localization_node" clear_params="true">
<param name="world_frame" value="odom"/>
<param name="odom0" value="/odometry/gps"/>
<param name="imu0" value="/kitti/oxts/imu"/>
<remap from="/odometry/filtered" to="/odometry/filtered_NAVI" />
</node>

<node pkg="robot_localization" type="navsat_transform_node" name="navsat_transform_node" respawn="true">
    <remap from="/imu/data" to="/kitti/oxts/imu" />
    <remap from="/gps/fix" to="/kitti/oxts/gps/fix" />
  </node>

</launch>

oconnor127, 2019-08-27:
I should add that the published odometry/filtered from the fusing imu node and the odometry/gps from navsat looks quite good... I assume that the odometry/gps are the fused gps and imu data?

ayrton04, 2019-08-27:
Would you mind posting questions on answers.ros.org? That way everyone gets the benefit of the answer. Thanks!

closed 2019-08-27 by oconnor127